RobotLocomotion/drake #2735: Java methods not found (MessageMonitor, BotLCMGLClient)

samkhal, 2016-07-03:
My drake install can't seem to find certain Java methods. I'm having a number of tests fail with messages like `Undefined variable "drake" or class "drake.util.BotLCMGLClient"` or `Undefined variable "drake" or class "drake.util.MessageMonitor"`. (i.e. test 84). The BotLCMGLClient.java file is present in /util/, but Drake can't seem to locate it.

I'm running R2016a or Ubuntu 14.04.

liangfok, 2016-07-04:
Is test 84 called "solvers/test/rrt_piano_mover_rotation"?

The problem sounds like the `CLASSPATH` environment variable is not being correctly set. Try searching for `BotLCMGLClient.class` and ensuring it is in your `CLASSPATH`. For example, here it is on my OS X machine:

```
$ find . -name "BotLCMGLClient*"
./drake/CMakeFiles/drake.dir/drake/util/BotLCMGLClient.class
```

In this case, I would try: 

```
$ cd drake/CMakeFiles/drake.dir/
$ export CLASSPATH=`pwd`:$CLASSPATH
```
If the above worked, it would at least determine what is the problem. It is not a long-term solution since the CLASSPATH should be set automatically by the unit testing framework.

FTR, I am also unable to run unit test "solvers/test/rrt_piano_mover_rotation", but I'm using OS X and am running into #2724.

samkhal, 2016-07-05:
Yes, that's the test I was talking about. I was able to get it working by setting the CLASSPATH, and in matlab by calling `javaaddpath('/home/user/drake-distro/drake/pod-build/CMakeFiles/drake.dir'). However, that gets cleared every time I restart matlab, so I'd have to put that in my startup script or call it every time. `addpath_drake` adds the .../share/java folders to the dynamic java path, but somehow that's not enough.

RussTedrake, 2016-09-08:
is this still an issue for you?  can you confirm that drake.jar specifically gets added to your dynamic classpath?  if drake.jar has a version issue, then it will fail to load -- silently (this is a terrible feature of matlab, in my opinion).

RussTedrake, 2016-09-16:
will close this until we hear more.

closed 2016-09-16 by RussTedrake

samkhal, 2016-09-16:
I haven't encountered this issue in a while; closing it was warranted.

hughhugh, 2017-06-03:
@RussTedrake 
Hi,
I have compiled the project Drake. In matlab examples like quadrotor, the runLQR works. The thing comes different when I am trying to run a simulation simflight, I always got errors like drake is not identified as a java class. I have tried the solution by @liangfok to deal with the error, but it does not work. I read the addpath_drake.m. and debug it. The dynamic java path is as the following after running addpath_drake.m.  Is it okay for setting environment for drake in Matlab? Thank you very much.
The system is ubuntu 16.04 with R2017a.
`>> javaclasspath('-dynamic')

		DYNAMIC JAVA PATH

	/home/name/drake_distro/drake/../build/install/matlab/../share/java/xmlgraphics-commons-1.3.1.jar
	/home/name/drake_distro/drake/../build/install/matlab/../share/java/lcmtypes_robotlocomotion.jar 
	/home/name/drake_distro/drake/../build/install/matlab/../share/java/lcmtypes_drake.jar           
	/home/name/drake_distro/drake/../build/install/matlab/../share/java/lcmtypes_bot2-procman.jar    
	/home/name/drake_distro/drake/../build/install/matlab/../share/java/lcmtypes_bot2-param.jar      
	/home/name/drake_distro/drake/../build/install/matlab/../share/java/lcmtypes_bot2-frames.jar     
	/home/name/drake_distro/drake/../build/install/matlab/../share/java/lcmtypes_bot2-core.jar       
	/home/name/drake_distro/drake/../build/install/matlab/../share/java/lcmspy_plugins_bot2.jar      
	/home/name/drake_distro/drake/../build/install/matlab/../share/java/lcm.jar                      
	/home/name/drake_distro/drake/../build/install/matlab/../share/java/jide-oss-2.9.7.jar           
	/home/name/drake_distro/drake/../build/install/matlab/../share/java/jchart2d-3.2.2.jar           
	/home/name/drake_distro/drake/../build/install/matlab/../share/java/drake.jar                    
	/home/name/drake_distro/drake/../build/install/matlab/../share/java/bot2-lcmgl.jar               
	/home/name/drake_distro/drake/../build/install/matlab/../share/java    `

RussTedrake, 2017-06-06:
Matlab's java bindings are frustrating.  Matlab will give an error like this if it (silently) failed to load the jar for any reason... incompatible java versions, etc.  It could be that lcm or some other jar was built without the proper java back-porting versions.  Best way to check might be to try instantiating classes from the supporting jar files.

Unfortunately, this workflow is deprecated now and is scheduled for deletion.  We're replacing it with better and more robust c++ interfaces.