RobotLocomotion/drake #2736: Clang-Mac CI Failure: No rule to make target: libpython2.7.dylib

liangfok, 2016-07-03:
# Example Log

Here is an example log:
https://drake-jenkins.csail.mit.edu/job/experimental/2464/compiler=clang,label=mac/consoleFull

# The Error

Here is an example error:

```
10:01:33 [  8%]make[9]: *** No rule to make target `/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/libpython2.7.dylib', needed by `lib/libvtkDRCFiltersPythonD.dylib'.  Stop.
```

# Root Cause Analysis

I haven't positively identified the root cause, but one thing I'm suspicious about is the fact that a different version of Python was found earlier in the same log:

```
10:01:28 -- Found PythonLibs: /usr/local/Cellar/python/2.7.12/Frameworks/Python.framework/Versions/2.7/lib/libpython2.7.dylib (found version "2.7.12") 
```

Note that the version found is version 2.7.12 whereas the version mentioned in the error message is 2.7.11.

sherm1, 2016-07-05:
Could someone at Kitware take a look at this persistent Mac failure (@jamiesnape?). Thanks!

jamiesnape, 2016-07-05:
Looking at it now.

jamiesnape, 2016-07-06:
An issue with an old VTK config file on the image. Now fixed. Unrelated to current OS X outage which is hardware related.

closed 2016-07-06 by jamiesnape