RobotLocomotion/drake #2739: Value iteration examples (Pendulum, Double Integrator) policy not working

samkhal, 2016-07-05:
The value iteration examples appear broken for me. Running `runValueIteration` in `drake/examples/Pendulum` produces roughly correct graphs for cost-to-go and policy, but the simulation is clearly wrong: it acts as if the input is constant. Inspecting the PI object shows a valid policy, but the feedback system `feedback(plant,PI)` acts as though the policy is a constant torque.

The same effect happens with DoubleIntegrator.runValueIteration, though there the resulting policy appears to be 0.

My drake install seems ok and most tests pass. Value iteration doesn't throw any errors. I remember these examples working fine when I took Underactuated at the end of 2015, so I'm wondering if it's something with my install or if the examples have stopped working from changes to Drake. Seems like something to do with the way feedback works with MarkovDecisionProcessPolicy systems.

vtjeng, 2016-10-24:
For `drake/examples/pendulum`, I obtained the following policy and cost-to-go plots, which seem roughly correct as @samkhal indicates.

[image]

However, rather than swinging all the way to the top as it used to, the pendulum appears to be stuck at approximately 30 degrees left of vertical. Will continue to investigate.

Running MATLAB R2016b on Ubuntu 14.04.

jwnimmer-tri, 2017-06-19:
Closing as won't-fix, because #6369 will be removing the relevant software.

If I'm confused and this should in fact stay open, please re-open an clarify the remaining unsolved problem(s) in a new comment here.

closed 2017-06-19 by jwnimmer-tri