RobotLocomotion/drake #2742: From source instructions: no mention of make options

tkoolen, 2016-07-05:
Why is there no longer any mention of `make options`?

jwnimmer-tri, 2016-07-05:
See #2555 where it was removed from documentation, and #2594 where it is planned to be removed from the Makefile as well.

From what I understand, the new advice is to run `ccmake` directly, instead of `make options`, though I suspect the documentation of that needs improvement.

I suspect that #2640 would make this all much clearer.

tkoolen, 2016-07-05:
There is currently one mention of ccmake, under the Ninja instructions, which I chose not to follow.

liangfok, 2016-07-06:
The documentation was updated in https://github.com/RobotLocomotion/drake/pull/2734, which was merged this morning. It'll probably take another day or so before the changes show up in the official http://drake.mit.edu website. Sorry for the confusion!

liangfok, 2016-07-10:
I'm going to close this issue because the following text now appears in http://drake.mit.edu/from_source.html under the subsection entitled "Build with `Make`":

> In addition to the build mode, Drake has many other build options. You can view them in drake-distro/CMakeLists.txt. The build options typically start with “WITH_”. The options can be included in the above cmake command by adding additional -D flags. You can also specify them using a text-based GUI by executing:
>
> > cd drake-distro/build
> > ccmake ../

closed 2016-07-10 by liangfok