RobotLocomotion/drake #2743: Build broken on GCC when Matlab is present

tkoolen, 2016-07-05:
Pull request https://github.com/RobotLocomotion/drake/pull/2717 disabled suppression of `no-sign-compare`. Together with `-Wall`, this now produces build errors in mex files for me:

```
/home/tkoolen/code/drake-distro/drake/systems/trajectories/shiftFunnel_snopt_mex.cpp: In function ‘double ptToPolyBullet(double*, std::size_t, std::size_t, mxArray*)’:
/home/tkoolen/code/drake-distro/drake/systems/trajectories/shiftFunnel_snopt_mex.cpp:156:21: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
   for (int i = 1; i < nCols; i++) {
                     ^
/home/tkoolen/code/drake-distro/drake/systems/trajectories/shiftFunnel_snopt_mex.cpp: In function ‘double* shiftAndTransform(double*, double*, const mxArray*, mxArray*, int, mxArray*, std::size_t, std::size_t)’:
/home/tkoolen/code/drake-distro/drake/systems/trajectories/shiftFunnel_snopt_mex.cpp:195:21: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
   for (int i = 0; i < nRows; i++) {
                     ^
/home/tkoolen/code/drake-distro/drake/systems/trajectories/shiftFunnel_snopt_mex.cpp:196:23: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
     for (int j = 0; j < nCols; j++) {
                       ^
/home/tkoolen/code/drake-distro/drake/systems/trajectories/shiftFunnel_snopt_mex.cpp: In function ‘bool penetrationCost(snopt::doublereal*, double*, double*)’:
/home/tkoolen/code/drake-distro/drake/systems/trajectories/shiftFunnel_snopt_mex.cpp:256:21: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
   for (int k = 0; k < N; k++) {
                     ^
/home/tkoolen/code/drake-distro/drake/systems/trajectories/replanFunnels_mex.cpp: In function ‘double ptToPolyBullet(double*, std::size_t, std::size_t, mxArray*)’:
/home/tkoolen/code/drake-distro/drake/systems/trajectories/replanFunnels_mex.cpp:102:21: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
   for (int i = 1; i < nCols; i++) {
                     ^
/home/tkoolen/code/drake-distro/drake/systems/trajectories/replanFunnels_mex.cpp: In function ‘double* shiftAndTransform(double*, double*, const mxArray*, mxArray*, int, mxArray*, std::size_t, std::size_t)’:
/home/tkoolen/code/drake-distro/drake/systems/trajectories/replanFunnels_mex.cpp:142:21: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
   for (int i = 0; i < nRows; i++) {
                     ^
/home/tkoolen/code/drake-distro/drake/systems/trajectories/replanFunnels_mex.cpp:143:23: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
     for (int j = 0; j < nCols; j++) {
                       ^
/home/tkoolen/code/drake-distro/drake/systems/trajectories/replanFunnels_mex.cpp: In function ‘bool penetrationCost(snopt::doublereal*, double*, double*)’:
/home/tkoolen/code/drake-distro/drake/systems/trajectories/replanFunnels_mex.cpp:274:21: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
   for (int k = 0; k < N; k++) {
                     ^
/home/tkoolen/code/drake-distro/drake/systems/trajectories/isCollisionFree_mex.cpp: In function ‘double ptToPolyBullet(double*, std::size_t, std::size_t)’:
/home/tkoolen/code/drake-distro/drake/systems/trajectories/isCollisionFree_mex.cpp:63:21: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
   for (int i = 1; i < nCols; i++) {
                     ^
/home/tkoolen/code/drake-distro/drake/systems/trajectories/replanFunnels_mex.cpp: In function ‘bool isCollisionFree(int, const mxArray*, const mxArray*, const mxArray*, mwSize, double*)’:
/home/tkoolen/code/drake-distro/drake/systems/trajectories/replanFunnels_mex.cpp:606:21: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
   for (int k = 0; k < N; k++) {
                     ^
/home/tkoolen/code/drake-distro/drake/systems/trajectories/isCollisionFree_mex.cpp: In function ‘double* shiftAndTransform(double*, double*, const mxArray*, mxArray*, int, mxArray*, std::size_t, std::size_t)’:
/home/tkoolen/code/drake-distro/drake/systems/trajectories/isCollisionFree_mex.cpp:97:21: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
   for (int i = 0; i < nRows; i++) {
                     ^
/home/tkoolen/code/drake-distro/drake/systems/trajectories/isCollisionFree_mex.cpp:98:23: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
     for (int j = 0; j < nCols; j++) {
                       ^
/home/tkoolen/code/drake-distro/drake/systems/trajectories/isCollisionFree_mex.cpp: In function ‘bool isCollisionFree(int, const mxArray*, const mxArray*, const mxArray*, mwSize, double*)’:
/home/tkoolen/code/drake-distro/drake/systems/trajectories/isCollisionFree_mex.cpp:136:21: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
   for (int k = 0; k < N; k++) {
                     ^
/home/tkoolen/code/drake-distro/drake/systems/trajectories/replanFunnels_mex.cpp: In function ‘void mexFunction(int, mxArray**, int, const mxArray**)’:
/home/tkoolen/code/drake-distro/drake/systems/trajectories/replanFunnels_mex.cpp:832:23: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
   for (int ii = 0; ii < numFunnels; ii++) {
                       ^
/home/tkoolen/code/drake-distro/drake/systems/trajectories/replanFunnels_mex.cpp:941:25: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
     for (int ii = 0; ii < numFunnels; ii++) {
                         ^
```

jwnimmer-tri, 2016-07-05:
Sorry for the trouble!

For the record, it was actually #2716 that un-disabled the warning in `systems/trajectories` (not #2717, which was for `systems/robotInterfaces`).

The Jenkins builds of master ("[Continuous](https://drake-jenkins.csail.mit.edu/view/Continuous/)") are 100% passing (including [linux-gcc-continuous-matlab](https://drake-jenkins.csail.mit.edu/view/Continuous/job/linux-gcc-continuous-matlab/)), so I'm not sure why the above errors are showing up.  I will look into it.

jwnimmer-tri, 2016-07-05:
As best I can tell, one of these conditions (from `drake-distro/drake/systems/trajectories/CMakeLists.txt:19`) is false in Jenkins:
```
if(BLAS_LIBRARY AND snopt_c_FOUND AND bullet_FOUND AND MATLAB_FOUND)
```
which means that Jenkins isn't covering all of the code.

I don't know how to read the cmake output to discover what's going on, but we can reason it out this way:
- The build log has `SnoptSolver.cpp.o` showing up, which means that `snopt_c_FOUND` must have been true.
- The build log has `bullet_model.cc.o` showing up, which means that `bullet_FOUND` must have been true.
- Matlab mex and tests are running, so `MATLAB_FOUND` must have been true.

That leaves `BLAS_LIBRARY` as the culprit, which is consistent with my expectations.

gizatt, 2016-07-05:
FWIW similar errors appear in the octomap wrapper if building with octomap.

jwnimmer-tri, 2016-07-05:
> FWIW similar errors appear in the octomap wrapper if building with octomap.

Filed as #2745.

jwnimmer-tri, 2016-07-05:
FTR, I am not working on this issue -- I don't have `snopt_c`, so I can't fix it locally, so Jekins is my only test case, and it's not building this code either.  I can review any PRs to fix this, but until Jenkins and/or the `CMakeLists.txt` are repaired so that this is covered in CI, it's not going to get fixed (or at least, it won't remain fixed).

liangfok, 2016-07-06:
Is `SNOPT_C` enabled by CMake option  `-DWITH_SNOPT=ON`?

jwnimmer-tri, 2016-07-06:
Yes, that is the CMake option to request that `SNOPT_C` be used.  (As with all `WITH_...` options, it is a polite request, not a guarantee).

liangfok, 2016-07-06:
Thanks. I am unable to reproduce the error due to #2723.

sammy-tri, 2016-07-06:
I'm hitting this on my machine.  Will attempt to fix.

PR #2753 (2016-07-06): Fix sign compare warnings from issue #2743.
This patch just makes the type of the loop counter variables match the
type being compared to.  While size_t is generally not favored in
Drake code, this is a MATLAB interface, and the function mxGetN() is
documented as returning
size_t. http://www.mathworks.com/help/matlab/apiref/mxgetn.html

Fixes #2743

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2753)
closed 2016-07-06 by jwnimmer-tri