RobotLocomotion/drake #2745: Build broken on GCC & Clang when Matlab and octomap are present

jwnimmer-tri, 2016-07-05:
Broken out from #2743, as reported by @gizatt.

On GCC & Clang builds, when Matlab and octomap are present, the `octomapWrapper.cpp` file is added to the build, but fails due to `-Werror=sign-compare` errors.

jwnimmer-tri, 2016-07-05:
@gizatt Can you provide more details?

According to [this build log](https://drake-jenkins.csail.mit.edu/view/Continuous/job/linux-gcc-continuous-matlab/413/consoleFull), `octomapWrapper.cpp` is being compiled by Jenkins, and is producing no warnings or errors.

gizatt, 2016-07-05:
I can't reproduce -- looking back at my console logs I can only guess I misread the output and attributed a build fail to the wrong thing. My bad! I'll keep an eye out for this resurging but I think it's safe.

closed 2016-07-05 by gizatt