RobotLocomotion/drake #2747: 2016-07-06 The mac builders seem to be offline

jwnimmer-tri, 2016-07-06:
The list:
```
mac-clang-ninja-nightly-matlab   	
mac-clang-ninja-nightly-matlab-open-source   	
mac-clang-nightly-matlab-open-source   	
mac-clang-nightly-matlab   	
mac-clang-nightly-debug   	
mac-clang-ninja-nightly-debug   	
mac-clang-nightly-release   	
mac-clang-ninja-nightly-release   	
mac-clang-continuous-release   	
mac-clang-ninja-continuous-release   	
mac-clang-ninja-continuous-debug   	
mac-clang-continuous-debug   	
experimental » clang,mac   	
experimental » clang,mac
```

Oldest says "Waiting for 14 hr".

jwnimmer-tri, 2016-07-06:
/cc @jhoare buildcop.

jamiesnape, 2016-07-06:
We have a hardware issue with the fiber channel.

jamiesnape, 2016-07-06:
We are on reduced capacity pending full resolution of the hardware issue.

jamiesnape, 2016-07-11:
OS X builds will be temporarily unavailable for around an hour.

jamiesnape, 2016-07-18:
OS X builds are available.

closed 2016-07-18 by jamiesnape