RobotLocomotion/drake #2754: Some systems/trajectories code isn't built by jenkins

jwnimmer-tri, 2016-07-06:
In #2743, we found a latent compile error that wasn't exposed by Jenkins.  It looks like some mex-related & blas-related & snopt-related & bullet-related trajectories code is only built conditionally, but for some reason the Jenkins builds aren't satisfying those conditions.

First, I would like to fix Jenkins so that it's covering this code.

Second, I would like to make it impossible to add built code to the tree unless the all-everything Jenkins built is covering it.  (Why isn't there a way to have the build fail, rather than downgrade its capabilities, when dependencies are unmet?)

jamiesnape, 2016-07-15:
Dependencies are not always properly declared, which is one problem.

> (Why isn't there a way to have the build fail, rather than downgrade its capabilities, when dependencies are unmet?)

Perfectly possible and already happens in many places.

jwnimmer-tri, 2016-09-19:
The lack of this CI coverage can make Russ's computer break.  Now high priority.

jamiesnape, 2016-09-19:
I guess maybe a `drake_find_package` that conditionally switches on `REQUIRED` for certain packages. Otherwise, you do want the downgrade behavior.

jwnimmer-tri, 2016-09-19:
Right.  If the user didn't mention the option, we can leave it as "best effort".  But if the user (or CI) demands the option up-front, we should fail-fast in its absence.

david-german-tri, 2016-10-06:
Remaining TODO here is to make sure that CI errors, instead of falling back to turning a package off, when finding the package fails.  (This would be good for developer builds too!)

May be facilitated by cleaning up package name inconsistencies.

PR #3718 (2016-10-21): Add and use drake_find_package macro
If a package is not `REQUIRED`, the find will be `QUIET` by default.  Moreover if `WITH_<PACKAGE>` is set to `ON` and <PACKAGE> is not found, then a fatal error will be thrown if `FIND_PACKAGE_STRICT` is `ON` (the default).

Fixes #2754.

----
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/3718)
closed 2016-10-21 by jamiesnape