RobotLocomotion/drake #2757: Improve developer workflow with submodules

bradking, 2016-07-07:
Since #2487 we use submodules for externals.  This has many advantages but also requires extra care in the developer workflow to avoid accidentally committing submodule revision changes.  I'm opening this issue to track the problems we encounter and discuss solutions.

bradking, 2016-07-07:
One problem was addressed in #2695.  PRs whose branches were started before that PR was merged may encounter problems with submodule updates.

bradking, 2016-07-07:
The fundamental issue is that `git pull` does not update the checked out submodules to match the version recorded in the `HEAD` commit of the outer project checked out after the pull.  A separate `git submodule update` command is needed for that.  We do this automatically as part of the build by default, but it is still possible for someone to `git commit -a` or `git add -u; git commit` without building.  If submodule checkouts were not updated then such commits will record the old version, effectively recording a rewind of submodules in the new commit.

bradking, 2016-07-07:
One approach to address this is to install a local developer Git commit hook that checks commit content when running `git commit`.  The hook can check for the case of a submodule rewind and reject it with an error message that explains what to do (and how to override it for those that really want to do it).

Git won't allow any actions to be performed by the project automatically after `git clone`, so we cannot make this hook installation fully automatic.  However, we could either document a developer setup script that does it, or have CMake install the hook on the first run.

bradking, 2016-07-07:
Cc: @billhoffman @mwoehlke-kitware

david-german-tri, 2016-07-13:
> However, we could either document a developer setup script that does it, or have CMake install the hook on the first run.

I would think the latter is easier to maintain across platforms, especially since the superbuild already runs a handful of custom git commands to deal with submodules.

david-german-tri, 2016-07-13:
@jwnimmer-tri also points out that the CMake approach avoids creating Yet Another Installation Step that Developers Can Get Wrong.

david-german-tri, 2016-08-18:
I think this was resolved in #2834.  I haven't heard any complaints from the team about submodules in several weeks!

sherm1, 2016-08-18:
> I haven't heard any complaints from the team about submodules in several weeks!

Well, I did lose another branch to mysterious "modified" submodule `spdlog` which would not revert. That was another case where the submodule was added to upstream after I started on that branch. When I pulled upstream to update my branch, after the merge `externals/spdlog` showed as modified locally. "revert" claimed success but had no effect. My ham-handed attempts to use `git submodule` didn't work and I believe I ended up damaging the index somehow. I decided just to abandon that branch and move to a clean one rather than complain about it.

david-german-tri, 2016-08-18:
The fix for all submodule problems is to re-run the superbuild.  Did you try that?  If the submodules still appear broken afterwards, that's a bug - please open an issue with the specifics, and assign it to me, @bradking, and @mwoehlke-kitware.

(Any build of Drake other than the superbuild is a shortcut for which there are no guarantees.)

sherm1, 2016-08-18:
Yes, I did run superbuild first but that left me with the alleged local modification. I've lost the specifics now but will report next time if I see it again.

bradking, 2016-08-23:
@sherm1 FYI the way to make local changes to a submodule is to `cd submodule_dir; git checkout $version_desired`.  Think of the submodule link in the outer project as a file containing the HEAD commit sha1 of the submodule.  The way to edit this "file" is to change the HEAD by checking out a different version in the submodule.

sherm1, 2016-08-23:
Thanks, @bradking. My problem so far has been that git reports modifications to a submodule dir that I didn't make, and there appears to be no way to revert.

bradking, 2016-08-24:
@sherm1 next time that happens please contact me via private email so we can work it out in detail.  I'd like to identify exactly what happens and determine how to fix it and perhaps prevent it from happening in the future.

sherm1, 2016-08-24:
@bradking, I think there is an example in #3202. In this case I had a branch I was working on, and had opened the PR. The branch got out of date with upstream so I pulled from there and merged, then pushed back to my fork. Now `externals/google_styleguide` shows in the file matrix as a commit in my PR :sad:. When I merge this PR won't that trash master's corresponding external?

bradking, 2016-08-25:
@sherm1 the appearance of `externals/google_styleguide` in the file matrix appears to be an artifact of how GitHub and Reviewable are presenting the changes.  They think the change was added to your topic since its original starting point.  However, the commit that makes the submodule update is already in `master` and so merging your topic will not touch it.

sherm1, 2016-08-25:
@bradking, thanks. I'm still worried about this, though. In #3202 the spurious commit wants to update the SHA to `f15e633de5b716a966504e0652fb0c85c2f4f4fb` but the one on master (as best I can tell) is `27929f1fd6e928ba536784e82ec67f7d474725c6`. So won't merging #3202 trash master's submodule?

bradking, 2016-08-25:
@sherm1 you're right.  The reason that `git log -- externals` doesn't show this is because it was the merge commit that caused the rewind so log assumes that the merge commit did not change the content because it matches at least one parent.  We can see that it took the submodule version from its first parent but we actually needed to keep the one from the second parent:

```console
$ git diff-tree --no-commit-id -r -m 6fe542a194523a7a260a8d06d0a949288ed4a0ac -- externals
:160000 160000 27929f1fd6e928ba536784e82ec67f7d474725c6 f15e633de5b716a966504e0652fb0c85c2f4f4fb M      externals/google_styleguide
$ git rev-parse 6fe542a194523a7a260a8d06d0a949288ed4a0ac:externals/google_styleguide                                                         
f15e633de5b716a966504e0652fb0c85c2f4f4fb
$ git rev-parse 6fe542a194523a7a260a8d06d0a949288ed4a0ac^1:externals/google_styleguide                                                       
f15e633de5b716a966504e0652fb0c85c2f4f4fb
$ git rev-parse 6fe542a194523a7a260a8d06d0a949288ed4a0ac^2:externals/google_styleguide                                                       
27929f1fd6e928ba536784e82ec67f7d474725c6
```

bradking, 2016-08-25:
@sherm1 I tried to reproduce your merge as follows:

```console
$ git checkout -b test-branch 69888e380373b91e06d9dda315caec8a0cf4975d
$ git submodule update
$ git branch test-master 3d877cbfc44210e9f93ec75b7fb0dea330f8d37e
$ git merge test-master
Auto-merging drake/systems/framework/system.h
CONFLICT (content): Merge conflict in drake/systems/framework/system.h
Auto-merging drake/systems/framework/primitives/constant_vector_source.cc
CONFLICT (content): Merge conflict in drake/systems/framework/primitives/constant_vector_source.cc
Auto-merging drake/systems/framework/primitives/adder.cc
Auto-merging drake/systems/framework/leaf_system.h
Auto-merging drake/systems/framework/diagram_context.h
Auto-merging drake/systems/framework/diagram.h
Auto-merging drake/systems/controllers/simple_feedback_trajectory_controller.h
Automatic merge failed; fix conflicts and then commit the result.
```

At this point I resolved the conflicts by taking the content from your original merge commit (for purposes of this test):

```console
$ git checkout 6fe542a194523a7a260a8d06d0a949288ed4a0ac -- drake/systems/framework/system.h drake/systems/framework/primitives/constant_vector_source.cc
```

However, one can see that the submodule has been left in the old state rather than that merged from `test-master`:

```console
$ git ls-files -s -- externals/google_styleguide/
160000 f15e633de5b716a966504e0652fb0c85c2f4f4fb 0       externals/google_styleguide
$ git rev-parse test-branch:externals/google_styleguide
f15e633de5b716a966504e0652fb0c85c2f4f4fb
$ git rev-parse test-master:externals/google_styleguide
27929f1fd6e928ba536784e82ec67f7d474725c6
```

I'm not sure why Git would do this; perhaps it forgot to update the submodule reference in the index after seeing a conflict.  Anyway, I'm still not able to create the commit due to the pre-commit hook we added:

```console
$ git commit
pre-commit hook failure
-----------------------

This commit would rewind a submodule link:

  "externals/google_styleguide"  27929f1 => f15e633

from the newer version in MERGE_HEAD (3d877cb).  Run

  git reset MERGE_HEAD -- "externals/google_styleguide"
  git submodule update -- "externals/google_styleguide"

to checkout the newer version of the submodule in your work tree.
Then try the commit again.

If you know what you are doing, run
  git config hooks."externals/google_styleguide".update "27929f1..f15e633"
to override this hook.
```

sherm1, 2016-08-25:
@bradking -- I still have no idea what happened but I was able to fix it using @liangfok's instructions in #3237. PR #3202 doesn't show that change now.

bradking, 2016-08-26:
Discussion in #3237 identified that #2896 (9147917ae050d3992fdb87769844d0c53551b065) broke the submodule-sync step for superbuilds not using the Ninja generator.  Re-opening.

PR #3270 (2016-08-27): Fix submodule sync dependencies
Fix setting up the dependencies on the global submodule sync target so that all projects with an update step depend on it. The old logic was only adding the dependency — to ALL projects — when Ninja is used. This latter is necessary due to quirks in how CMake computes dependencies for Ninja (fixed in 3.7), but the former, which should be done for obvious reasons, was missing. (This got messed up setting up the dependencies for the Ninja case.)

Fixes #2757.

Review: @bradking (feature), @david-german-tri (platform)

---
This change is [[image]](https://reviewabl [... cut]
closed 2016-08-27 by david-german-tri