RobotLocomotion/drake #2765: Problem to launch drake-visualizer

hongkai-dai, 2016-07-10:
The problem is partly described in #2731. After installing a fresh drake, I have problem to launch drake-visualizer from terminal. When I call 
```
DRAKE_DISTRO_FOLDER/build/install/bin/drake-visualizer
```
I get the following error
```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/hongkai/drake-distro/build/install/lib/python2.7/dist-packages/director/drakevisualizer.py", line 3, in <module>
    from director import lcmUtils
  File "/home/hongkai/drake-distro/build/install/lib/python2.7/dist-packages/director/lcmUtils.py", line 1, in <module>
    import lcm
  File "/home/hongkai/drake-distro/build/install/lib/python2.7/site-packages/lcm/__init__.py", line 3, in <module>
    from ._lcm import LCM, LCMSubscription
ImportError: dynamic module does not define init function (init_lcm)
```
`drake-distro/build/install/lib/python2.7/site-packages/lcm` folder contains three files, `__init__.py, __init__.pyc` and `_lcm.so`. In `_lcm.so`, it does not contain the symbol `init_lcm`. I did
```
hongkai@jiahui-All-Series:~/drake-distro/build/install/lib/python2.7/site-packages/lcm% ls
__init__.py  __init__.pyc  _lcm.so
hongkai@jiahui-All-Series:~/drake-distro/build/install/lib/python2.7/site-packages/lcm% ls -lah _lcm.so
-rw-r--r-- 1 hongkai hongkai 66K Jul  8 22:06 _lcm.so
hongkai@jiahui-All-Series:~/drake-distro/build/install/lib/python2.7/site-packages/lcm% nm -gC _lcm.so |grep init_lcm
hongkai@jiahui-All-Series:~/drake-distro/build/install/lib/python2.7/site-packages/lcm% 
```
The LCM version is 8ffd4d4, the drake-distro is at 29901cb

liangfok, 2016-07-10:
It's interesting that your `_lcm.so` is only 66KB whereas mine is 88KB. 

I am using the same version of LCM version:

```
$ cd drake-distro/.git/modules/externals/lcm
$ git rev-parse HEAD
8ffd4d4f5dfeab7a632b6e0517365ef8b848dc34
```

However, I am using a newer version of Drake:

```
$ cd drake-distro
$ git rev-parse HEAD
215e18fe3fd1d04e4da7dccca8bb2efe4c430dc1
```

Have you tried using the latest version of Drake (SHA 215e18fe3fd1d04e4da7dccca8bb2efe4c430dc1)?

Here is the verbose build output of build target `lcm-python`, which generates the `_lcm.so` file:

```
make -f lcm-python/CMakeFiles/lcm-python.dir/build.make lcm-python/CMakeFiles/lcm-python.dir/depend
make[5]: Entering directory `/home/liang/dev/drake-distro-1/build/externals/lcm'
cd /home/liang/dev/drake-distro-1/build/externals/lcm && /home/liang/tools/cmake-3.5.2-Linux-x86_64/bin/cmake -E cmake_depends "Unix Makefiles" /home/liang/dev/drake-distro-1/externals/lcm /home/liang/dev/drake-distro-1/externals/lcm/lcm-python /home/liang/dev/drake-distro-1/build/externals/lcm /home/liang/dev/drake-distro-1/build/externals/lcm/lcm-python /home/liang/dev/drake-distro-1/build/externals/lcm/lcm-python/CMakeFiles/lcm-python.dir/DependInfo.cmake --color=
Dependee "/home/liang/dev/drake-distro-1/build/externals/lcm/lcm-python/CMakeFiles/lcm-python.dir/DependInfo.cmake" is newer than depender "/home/liang/dev/drake-distro-1/build/externals/lcm/lcm-python/CMakeFiles/lcm-python.dir/depend.internal".
Dependee "/home/liang/dev/drake-distro-1/build/externals/lcm/lcm-python/CMakeFiles/CMakeDirectoryInformation.cmake" is newer than depender "/home/liang/dev/drake-distro-1/build/externals/lcm/lcm-python/CMakeFiles/lcm-python.dir/depend.internal".
Scanning dependencies of target lcm-python
make[5]: Leaving directory `/home/liang/dev/drake-distro-1/build/externals/lcm'
make -f lcm-python/CMakeFiles/lcm-python.dir/build.make lcm-python/CMakeFiles/lcm-python.dir/build
make[5]: Entering directory `/home/liang/dev/drake-distro-1/build/externals/lcm'
[ 81%] Building C object lcm-python/CMakeFiles/lcm-python.dir/module.c.o
cd /home/liang/dev/drake-distro-1/build/externals/lcm/lcm-python && /usr/bin/gcc-4.9  -Dlcm_python_EXPORTS -I/home/liang/dev/drake-distro-1/build/externals/lcm/lcm-python -I/home/liang/dev/drake-distro-1/externals/lcm/lcm-python -I/usr/include/python2.7 -I/home/liang/dev/drake-distro-1/externals/lcm  -std=gnu99 -Wno-format-zero-length -O2 -g -DNDEBUG -fPIC   -o CMakeFiles/lcm-python.dir/module.c.o   -c /home/liang/dev/drake-distro-1/externals/lcm/lcm-python/module.c
[ 83%] Building C object lcm-python/CMakeFiles/lcm-python.dir/pyeventlog.c.o
cd /home/liang/dev/drake-distro-1/build/externals/lcm/lcm-python && /usr/bin/gcc-4.9  -Dlcm_python_EXPORTS -I/home/liang/dev/drake-distro-1/build/externals/lcm/lcm-python -I/home/liang/dev/drake-distro-1/externals/lcm/lcm-python -I/usr/include/python2.7 -I/home/liang/dev/drake-distro-1/externals/lcm  -std=gnu99 -Wno-format-zero-length -O2 -g -DNDEBUG -fPIC   -o CMakeFiles/lcm-python.dir/pyeventlog.c.o   -c /home/liang/dev/drake-distro-1/externals/lcm/lcm-python/pyeventlog.c
[ 84%] Building C object lcm-python/CMakeFiles/lcm-python.dir/pylcm.c.o
cd /home/liang/dev/drake-distro-1/build/externals/lcm/lcm-python && /usr/bin/gcc-4.9  -Dlcm_python_EXPORTS -I/home/liang/dev/drake-distro-1/build/externals/lcm/lcm-python -I/home/liang/dev/drake-distro-1/externals/lcm/lcm-python -I/usr/include/python2.7 -I/home/liang/dev/drake-distro-1/externals/lcm  -std=gnu99 -Wno-format-zero-length -O2 -g -DNDEBUG -fPIC   -o CMakeFiles/lcm-python.dir/pylcm.c.o   -c /home/liang/dev/drake-distro-1/externals/lcm/lcm-python/pylcm.c
[ 86%] Building C object lcm-python/CMakeFiles/lcm-python.dir/pylcm_subscription.c.o
cd /home/liang/dev/drake-distro-1/build/externals/lcm/lcm-python && /usr/bin/gcc-4.9  -Dlcm_python_EXPORTS -I/home/liang/dev/drake-distro-1/build/externals/lcm/lcm-python -I/home/liang/dev/drake-distro-1/externals/lcm/lcm-python -I/usr/include/python2.7 -I/home/liang/dev/drake-distro-1/externals/lcm  -std=gnu99 -Wno-format-zero-length -O2 -g -DNDEBUG -fPIC   -o CMakeFiles/lcm-python.dir/pylcm_subscription.c.o   -c /home/liang/dev/drake-distro-1/externals/lcm/lcm-python/pylcm_subscription.c
[ 87%] Linking C shared module ../lib/python2.7/site-packages/lcm/_lcm.so
cd /home/liang/dev/drake-distro-1/build/externals/lcm/lcm-python && /home/liang/tools/cmake-3.5.2-Linux-x86_64/bin/cmake -E cmake_link_script CMakeFiles/lcm-python.dir/link.txt --verbose=TRUE
/usr/bin/gcc-4.9  -fPIC  -std=gnu99 -Wno-format-zero-length -O2 -g -DNDEBUG  -shared  -o ../lib/python2.7/site-packages/lcm/_lcm.so CMakeFiles/lcm-python.dir/module.c.o CMakeFiles/lcm-python.dir/pyeventlog.c.o CMakeFiles/lcm-python.dir/pylcm.c.o CMakeFiles/lcm-python.dir/pylcm_subscription.c.o ../lcm/liblcm.so.1.3.1 -lpython2.7 -Wl,-rpath,/home/liang/dev/drake-distro-1/build/externals/lcm/lcm: 
make[5]: Leaving directory `/home/liang/dev/drake-distro-1/build/externals/lcm'
[ 87%] Built target lcm-python
```
How does it compare to yours? Below are the commands I executed to get the above output:

```
$ mkdir build
$ cd build
$ cmake ../ -DCMAKE_BUILD_TYPE:STRING=RelWithDebInfo -DWITH_SNOPT=TRUE -DWITH_SNOPT_PRECOMPILED=FALSE
$ make -j1 VERBOSE=TRUE 2>&1 | tee -a buildlog.txt
```

liangfok, 2016-07-10:
On my computer, the `init_lcm` symbol is coming from the object file resulting from compiling `drake-distro/externals/lcm/lcm-python/module.c`:

```
$ cd drake-distro/build
$ nm -gC `find . -name module.c.o` | grep init_lcm
0000000000000000 T init_lcm
```

Is your `module.c.o` missing this symbol?

hongkai-dai, 2016-07-10:
@liangfok , thanks a lot for your help! my `module.c.o` does not have the `init_lcm` symbol. Here is what I did
```
hongkai@jiahui-All-Series:~/drake-distro/build/externals/lcm/lcm-python/CMakeFiles/lcm-python.dir% nm -gC module.c.o
                 U _GLOBAL_OFFSET_TABLE_
0000000000000000 T PyInit__lcm
                 U pylcmeventlog_type
                 U pylcm_subscription_type
                 U pylcm_type
                 U PyModule_AddObject
                 U PyModule_Create2
                 U PyType_Type

```
The build log for lcm is very similar to yours, but mine is using python3.4m instead of python 2.7. I suspect that is the reason. Attached is the diff file between our build logs (I replaced the directory name with mine to reduce the mismatches)
[buildlog_diff.txt](https://github.com/RobotLocomotion/drake/files/356096/buildlog_diff.txt)

hongkai-dai, 2016-07-10:
Just realized in #2731 , @rdeits has already mentioned this is a python version problem

hongkai-dai, 2016-07-10:
In `drake-distro/build/externals/lcm.CMakeCache.txt`, it sets the python library to python 3.4
```
//Path to a file.
PYTHON_INCLUDE_DIR:PATH=/usr/include/python3.4m

//Path to a library.
PYTHON_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libpython3.4m.so
```
I changed the `CMakeCache.txt` file to set these libraries to python2.7, and then my director can work. Is there a way set these variables during configuring? I tried to set `PYTHON_INCLUDE_DIR` AND `PYTHON_LIBRARY` when I call `cmake ../ -DPYTHON_INCLUDE_DIR:PATH=/usr/include/python2.7`, but cmake tells me that this variable is not used.

liangfok, 2016-07-10:
I'm glad you identified the problem! This is a complete guess but maybe try changing the first line of `drake-distro/externals/lcm/lcm-python/CMakeLists.txt` from:

```
find_package(PythonInterp REQUIRED)
```

to be:

```
find_package(PythonInterp 2.7 REQUIRED)
```

rdeits, 2016-07-10:
@liangfok that should work in this case, but it will break the LCM python bindings for other users who actually do want to use python 3. We may need to add a compilation option to LCM to select a python version (which I've already done for Drake: https://github.com/RobotLocomotion/drake/blob/master/drake/CMakeLists.txt#L44 )

closed 2016-07-29 by hongkai-dai