RobotLocomotion/drake #2767: drake-visualizer needs test coverage in CI

RussTedrake, 2016-07-10:
https://github.com/RobotLocomotion/drake/issues/2731 should have been caught immediately by CI -- drake-visualizer simply failed to run on homebrew mac.  We do not currently have any test coverage for running the drake-visualizer.

david-german-tri, 2016-07-11:
If I'm understanding correctly, this issue is asking for regression tests that run the whole `drake-visualizer` executable.  To do this right, we'd need a framework that allows us to make assertions on the graphical output.  However, if we just want to test that `drake-visualizer` starts up, we could modify it to emit an "I'm ready" message.

liangfok, 2016-07-11:
I believe that for now just verifying that `drake-visualizer` starts up would be sufficient since that would have caught #2731. Another option to having it emit an "I'm ready" message would be to add a command line argument that causes it to start up but then immediately shut down with a special exit code that can then be checked.

wxmerkt, 2016-07-11:
@david-german-tri director has some basic tests itself already, cf. https://github.com/RobotLocomotion/director/blob/master/src/python/tests/CMakeLists.txt

I believe adding one that will start https://github.com/RobotLocomotion/director/blob/master/src/python/director/drakevisualizer.py using directorPython and running the basic director test suite as part of the drake CI coverage might do the trick (with a timer to quit after 5 seconds).

david-german-tri, 2016-07-14:
@wxmerkt Thanks for the pointer!  Taking a look now.

david-german-tri, 2016-07-14:
When I run `ctest -VV` in `drake-distro/externals/director/distro/pods/drake-distro/pod-build/src/director-build/src/python/tests`, most of the tests fail due to missing `import`s. The reason is that the Drake superbuild doesn't build all of Director.  We probably want to keep it that way.  So, I'm going to explore a black-box test instead.

https://github.com/RobotLocomotion/drake/blob/master/CMakeLists.txt#L167

david-german-tri, 2016-07-15:
As an additional wrinkle, it turns out `drake-visualizer` and `directorPython` understandably will not start up without an X server.

jwnimmer-tri, 2016-07-15:
Does http://manpages.ubuntu.com/manpages/trusty/man1/xvfb-run.1.html help?

david-german-tri, 2016-07-15:
Interesting.  Perhaps so, if I can figure out the right incantation to turn on GLX.

mwoehlke-kitware, 2016-07-15:
> When I run `ctest -VV` in `drake-distro/externals/director/distro/pods/drake-distro/pod-build/src/director-build/src/python/tests`, most of the tests fail due to missing `import`s. The reason is that the Drake superbuild doesn't build all of Director.

Why is director emitting tests for stuff that wasn't built?

david-german-tri, 2016-07-15:
It looks like `xvfb-run` works just fine with GLX on AWS.  (Locally, it doesn't play nicely with my NVIDIA driver.)  So, my high-level plan is:

1. Write a Python script that runs drake-visualizer for a few seconds.  I have yet to find an approach that lets me send a graceful quit signal to the `DrakeVisualizerApp`.  Worst-case, I don't actually need to start the app, in the manner of https://github.com/RobotLocomotion/director/blob/master/src/python/tests/testDrakeVisualizer.py

2. Run that script using `directorPython` in a CMake `add_test` clause.  On platforms where a display is present, that just works.  If there is no display, we need `xvfb-run`.  That entails some subtasks:

2a. In CMake, detect whether a display is present or not.  Anyone have ideas about how to do that?
2b. Add Xvfb to the Linux CI images.
2c. Confirm there's a usable display in the Mac CI context.

(There will be no coverage for Windows because Drake doesn't yet build director on Windows.)

david-german-tri, 2016-07-15:
> Why is director emitting tests for stuff that wasn't built?

If I'm understanding it correctly, the directory containing the tests is built, but not all of the dependencies for all of the python code under test are built.  `import` errors ensue.

jwnimmer-tri, 2016-07-15:
> On platforms where a display is present, that just works. If there is no display, we need xvfb-run. That entails some subtasks:
> 2a. In CMake, detect whether a display is present or not. Anyone have ideas about how to do that?

Perhaps: in the test, if there is no X, then `subprocess.check_call()` ourself recursively with xvfb-run stuck first.  That way the user experience is just run the test, and it sets up stubs as needed.

mwoehlke-kitware, 2016-07-15:
> the directory containing the tests is built, but not all of the dependencies for all of the python code under test are built

Let me rephrase: is there a reason why director calls `add_test` for tests with optional dependencies without checking if the dependencies are enabled?

mwoehlke-kitware, 2016-07-15:
> In CMake, detect whether a display is present or not. Anyone have ideas about how to do that?

Check if `$DISPLAY` is set?

TBH, I'd consider just always using a VFB, to avoid obnoxiously popping up windows. Maybe as an option (default=ON) in case of systems where xvfb doesn't work for whatever reason.

rdeits, 2016-07-15:
Pat reports successfully using xvfb with director in the past: #1256

PR #2917 (2016-07-29): Add rudimentary test coverage for drake-visualizer.
  This test confirms that drake-visualizer will start up.  Since the Linux CI instances do not have Xvfb yet, and Windows does not have Director, this test only covers OS X.

Resolves #2767.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2917)
closed 2016-07-29 by david-german-tri