RobotLocomotion/drake #2771: CI Failure - gcc-linux-matlab - cddliv download error

liangfok, 2016-07-11:
# Example Logs

https://drake-jenkins.csail.mit.edu/job/experimental/2572/compiler=gcc,label=linux_matlab/consoleFull

# The Error

```
21:13:33 [ 11%] Performing download step (download, verify and extract) for 'cdd_project'
21:13:33 -- downloading...
21:13:33      src='ftp://ftp.ifor.math.ethz.ch/pub/fukuda/cdd/cddlib-094h.tar.gz'
21:13:33      dst='/home/ubuntu/workspace/experimental/06107e1d/externals/iris/pod-build/cdd_project-prefix/src/cddlib-094h.tar.gz'
21:13:33      timeout='none'
21:13:33 CMake Error at cdd_project-stamp/download-cdd_project.cmake:21 (message):
21:13:33   error: downloading
21:13:33   'ftp://ftp.ifor.math.ethz.ch/pub/fukuda/cdd/cddlib-094h.tar.gz' failed
21:13:33 
21:13:33     status_code: 7
21:13:33     status_string: "Couldn't connect to server"
21:13:33     log:   Trying 129.132.145.45...
21:13:33 
21:13:33   connect to 129.132.145.45 port 21 failed: Connection timed out
21:13:33 
21:13:33   Failed to connect to ftp.ifor.math.ethz.ch port 21: Connection timed out
21:13:33 
21:13:33   Closing connection 0
21:13:33 
```

jamiesnape, 2016-07-18:
The ETH server is just very, very slow. Not CI specific.

jwnimmer-tri, 2016-08-17:
Possibly we should not be downloading a tarball from Zurich hundreds of times per day?  Does its license allow re-hosting?

jwnimmer-tri, 2016-08-17:
License is GPL2+, so yes.  The https://launchpad.net/ubuntu/+archive/primary/+files/cddlib_094h.orig.tar.gz seems strictly superior to the overloaded FTP server, though hosting it on RobotLocomotion github would be even better.

jamiesnape, 2016-08-17:
Or it can go in an s3 bucket with some of our other dependencies. Either way, we should open a ticket with iris, I think.

jwnimmer-tri, 2016-08-17:
New issue filed.  I didn't realize cddlib was coming indirectly from Iris, instead of as a direct external of Drake.

jwnimmer-tri, 2016-08-17:
@jamiesnape I don't have a strong preference on how to host it, though I slightly prefer something we own instead of relying on Ubuntu or etc.  Would you prefer to add this tarball to S3 as the best choice?  If kitware can get that URL set up, I can prepare a PR to `iris-distro` with the change (or kitware can do it all, doesn't matter to me).

sammy-tri, 2016-08-17:
FWIW, this was a significant driver behind creating the local RobotLocomotion fork/mirror of IPOPT.  It also wanted to download dependencies from not-totally-reliable servers on the other side of an ocean, which are checked into the local fork/mirror instead.  (I think I worded that really poorly somehow, but hopefully communicated the idea...)

jamiesnape, 2016-08-17:
https://s3.amazonaws.com/drake-provisioning/cddlib-094h.tar.gz
SHA256: 6330b766690da30991c207ab75c96f7f57813273ee894e78d78506bebc6184c4

jwnimmer-tri, 2016-08-17:
PR filed as rdeits/iris-distro#42.

jwnimmer-tri, 2016-08-17:
@RussTedrake It looks to me like `IRIS` can only be enabled if you have `MOSEK`, which TRI doesn't have access to.  Can you assign someone to figure out why #3172's solution to a new `cddlib` URL isn't working?  We have no way to test it locally.  We are turning off `IRIS` in the meantime.  Thanks!

jwnimmer-tri, 2016-08-18:
Tentatively fixed by #3172 and wxmerkt finding an old copy of the required tarball to replace our somehow-wrong-copy in the S3 bucket.

PR #3172 (2016-08-18): Update iris to latest master
Fixes #2771.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/3172)
closed 2016-08-18 by david-german-tri