RobotLocomotion/drake #2772: lcm java interface to matlab is broken (at least on mac R2015b)

RussTedrake, 2016-07-11:
My version of matlab on mac is no longer able to load and use the lcm java bindings in lcm.jar.  To reproduce, simply run
```
addpath_drake
checkDependency('lcm')
```
in matlab.  

Once again, this is sneaking through our CI -- probably because missing dependencies currently fail silently.  This means that the matlab functionality in drake is absolutely crippled (It is also blocking the work I was hoping to do today.)

Having not looked into it yet, i suspect that flags like supporting older java versions (which have been the subject of conversation before) are missing.
```
>> version -java

ans =

Java 1.7.0_75-b13 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
```
but at the command line
```
drake008% javac -version
javac 1.8.0_60
```

david-german-tri, 2016-07-11:
Possibly related: #2333

jamiesnape, 2016-07-11:
I strongly recommend installing 1.7 (I thought we added that to the docs, but apparently not). Otherwise, you may set `CMAKE_JAVA_COMPILE_FLAGS` to `-source 1.7 -target 1.7` and proceed with caution.

david-german-tri, 2016-07-11:
@jamiesnape Doesn't #2527 do that automatically?

jamiesnape, 2016-07-11:
So it does. Maybe this is related to the refactoring of LCM.

FYI @mwoehlke-kitware.

david-german-tri, 2016-07-11:
In fact, I can reproduce this problem on Ubuntu 14.04 with Java 7, so I suspect the version mismatch is a red herring.

david-german-tri, 2016-07-11:
After running `addpath_drake`, no Drake-specific files or directories appear in my `javaclasspath`.

liangfok, 2016-07-11:
This PR could be related: https://github.com/RobotLocomotion/drake/pull/2599.

david-german-tri, 2016-07-11:
@liangfok Yes, possibly.  It looks like this line produces the wrong results for an out-of-source build, even with #2599.  I may not be reproducing the original problem after all.

https://github.com/RobotLocomotion/drake/blob/master/drake/addpath_drake.m#L100

@RussTedrake, could you post your `javaclasspath`?

RussTedrake, 2016-07-11:
it looks like #2527 sets things up correctly for the drake subdirectory, but not for the externals.  what if we moved that logic up to the superbuild directory, and make sure that lcm also uses the updated `CMAKE_JAVA_COMPILE_FLAGS`?

RussTedrake, 2016-07-11:
fwiw - i think a bigger issue here is that there are too many big changes happening (eg. refactoring LCM) for things that we don't have sufficient CI coverage on.

jamiesnape, 2016-07-11:
True, but the build system changes will be a huge improvement and it is difficult to test the many different versions of operating systems, Java, and MATLAB.

RussTedrake, 2016-07-11:
both of these were stock mac/homebrew issues -- that would have appeared on the current build systems if we had a test for them.

jamiesnape, 2016-07-11:
There is no stock Java on OS X any more. CI is OS X Yosemite with Java 1.7.

mwoehlke-kitware, 2016-07-11:
> it looks like #2527 sets things up correctly for the drake subdirectory, but not for the externals. what if we moved that logic up to the superbuild directory, and make sure that lcm also uses the updated `CMAKE_JAVA_COMPILE_FLAGS`?

That seems reasonable. It should suffice to pass `CMAKE_JAVA_COMPILE_FLAGS` into the externals; I don't think the externals themselves need to change. I'd make this change to `COMMON_CMAKE_ARGS`. (Alas, this is going to conflict with #2758 :cry:.)

RussTedrake, 2016-07-11:
thanks @mwoehlke-kitware -- can you make the PR?

mwoehlke-kitware, 2016-07-11:
> can you make the PR?

Added to #2758. The matlab build for that is breaking for mysterious reasons; textbook seems to be... missing. I can't reproduce on either my local machine nor on an AWS instance without matlab. I also can't test matlab because I don't seem to know the magic invocation for the license server; will try again tomorrow when folks are back in the office.

closed 2016-07-22 by mwoehlke-kitware

wxmerkt, 2016-10-17:
@mwoehlke-kitware This issue resurfaced after the LCM CMake upgrade with the current LCM master tip installed system-wide - is it possible to port [``drake_setup_java_for_matlab``](https://github.com/RobotLocomotion/drake/blob/137d3995fc30da0016ef4b2b6a0c80f1129cfe61/cmake/config.cmake#L59) over to LCM as well? A very common workflow which is now broken is to have LCM installed system-wide

mwoehlke-kitware, 2016-10-17:
> is it possible to port drake_setup_java_for_matlab over to LCM as well?

Probably, but that doesn't strike me as TRTTD. Wouldn't you need to do this for *any* distro-provided or built-outside-of-drake Java package that might be used by MATLAB? Why should upstream LCM (or any other project) care if you happen to have MATLAB installed on your system?

wxmerkt, 2016-10-17:
I've opened an upstream PR to fix the issue by setting source and target to 1.6 in any case - this mimicks what autoconfigure used to do (but back then setting it to 1.5) and works just fine - cf. https://github.com/lcm-proj/lcm/pull/137

PR #2820 (2016-07-22): Superbuild refactor, part 1
Remove (superbuild level) PODS support. Move general build configuration logic to a separate file and refactor to share this between the superbuild and the build of drake itself. Move logic to set the Java target version to match the matlab JVM into here, and pass Java build flags to all (CMake) externals. Improve compiler version checks and fix message for MSVC version.

Fixes #2772.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2820)