RobotLocomotion/drake #2786: Some functions in drake/math implicitly convert scalar types to doubles

david-german-tri, 2016-07-12:
See discussion on #2779.

sherm1, 2016-07-12:
To clarify, there are several math functions included in #2779's commits that are templatized by `Scalar` but use `double` intermediate calculations. It appears that those would produce incorrect derivatives if templatized by `AutoDiffScalar` because some of the chain rule pieces would be missing. However, there are no comments in that code and it is difficult to discern whether the switch to `double` was intentional so each method will have to be scrutinized carefully, preferably armed with good unit tests for the underlying calculation and derivative.

david-german-tri, 2017-04-13:
Reassigning to @hongkai-dai, who appears to be the de facto owner of `drake/math`.

jwnimmer-tri, 2019-03-30:
(1) Paul Mitiguy rewrote most of the math primitives since this issue was filed.
(2) An assignment from AutoDiffScalar to double will fail to compile.  You need to call `.value()` for it to work, so there wasn't really any lingering hazard in the first place.

closed 2019-03-30 by jwnimmer-tri