RobotLocomotion/drake #2788: arrow3D and ode4 are unlicensed

jwnimmer-tri, 2016-07-12:
Drake distributes the files `drake/thirdParty/nonfree/arrow3D.m` and `drake/thirdParty/nonfree/ode4.m`, but neither file appears to come with a license grant.  Unless we can obtain a such a license, we will have to stop using these two files.

arrow3D is used here:
- examples/grasping/forceClosure/ForceClosureContacts.m:138
- examples/grasping/forceClosure/ForceClosureContactsFixedFaces.m:94
- solvers/trajectoryOptimization/ContactWrenchVisualizer.m:138
- systems/robotInterfaces/FootstepPlan.m:120

ode4 is used here:
- examples/ZMP/LinearInvertedPendulum.m:291
- examples/dev/SpringFlamingo/SpringFlamingoPlant.m:7
- examples/dev/SpringFlamingo/SpringFlamingoPlant.m:48

arrow3D appears to be sourced from here, but note that this post has no license granted:
http://www.mathworks.com/matlabcentral/fileexchange/12274-3d-arrow-with-many-color-parameter-options/content/arrow3D_pub/arrow3D.m

ode4 appears to be sourced from here, but I can't find any copyright or license information:
http://www.nt.ntnu.no/users/skoge/diplom/diplom12/hansen/OLGA_Model_Simulations/ode4.m

jwnimmer-tri, 2016-07-12:
@RussTedrake seeking your thoughts on the best approach here.

hongkai-dai, 2016-07-12:
Sorry I added `arrow3D` into the code base, without realizing we violate the license rules. I believe `ContactWrenchVisualizer` and `FootstepPlan` are not using `arrow3D`, instead they use the `drawArrow3d` function from LCMGL. If using `arrow3D` is an issue, I can modify the force closure grasping code

hongkai-dai, 2016-07-12:
We can use `arrow3d` function from https://www.mathworks.com/matlabcentral/fileexchange/28324-3d-arrow-plot/content/arrow3d.m, it does have a BSD license.

jwnimmer-tri, 2016-07-12:
> I believe ContactWrenchVisualizer and FootstepPlan are not using arrow3D, instead they use the drawArrow3d function from LCMGL.

Correct, sorry!  Case-insensitive grep misled me.

jwnimmer-tri, 2016-07-12:
> We can use arrow3d function from https://www.mathworks.com/matlabcentral/fileexchange/28324-3d-arrow-plot/content/arrow3d.m, it does have a BSD license.

Sounds good to me, though I don't have the ability to test that change, so I'll need a volunteer to create the PR.

hongkai-dai, 2016-07-12:
I can create a PR to update the call to `arrow3D` functions

jwnimmer-tri, 2016-07-13:
For the record: ode4 is the only remaining problem now.

RussTedrake, 2016-07-14:
The only use of ode4 is in examples/ZMP/LinearInvertedPendulum.m .  @hongkai-dai, want to help purge that one, too?

hongkai-dai, 2016-07-14:
I find an ode4 package here http://www.mathworks.com/matlabcentral/fileexchange/54611-solving-odes-in-matlab, it comes with Mathworks Limited license, and requires MATLAB R2014b or later versions. Is this license compatible with Drake? Also, do we want to restrict to the newer version (>R2014b) of MATLAB?

Alternatively, I can write an ode4.m file by myself.

jwnimmer-tri, 2016-07-14:
I took a quick look at the Mathworks Limited license, but wasn't able to come to an immediate conclusion.  It may be beyond my lawyercat abilities, but I will give it more of my attention tomorrow and see if I can offer an answer.

avalenzu, 2016-07-14:
Maybe I'm missing something, but why do we need a non-adaptive integrator here?

RussTedrake, 2016-07-15:
Probably because I wanted the explicit solution to have an exact representation of the numerical solution (piecewise polynomial with known breaks)?  But definitely worth revisiting.

hongkai-dai, 2016-07-15:
In `SpringFlamingoPlant.m` , when we simulate it without Jerry's controller, it is probably not using `ode4` in line 7,8, notice line 7 (the first line below) is commented out
```
%      obj = setSimulinkParam(obj,'Solver','ode4','FixedStep','.0001');
       obj = setSimulinkParam(obj,'MinStep','0.0001');
```
When we simulate it with Jerry's controller, we require `ode4` as the solver. Is Jerry's controller a discrete time controller?

hongkai-dai, 2016-07-15:
@RussTedrake , in `LinearInvertedPendulum.m`, we call `ode4` to compute `y` and `ydot` as
```
[t,y,ydot] = ode4(@s2dynamics,fliplr(breaks),0);
```
and then compute `s2traj` as a cubic spline
```
s2traj = PPTrajectory(pchipDeriv(breaks,fliplr(y.'),fliplr(ydot.')));
```
I do not think the cubic spline matches with the 4th order Runge-Kutta method exactly. So maybe using `ode4` here is not necessary?

RussTedrake, 2016-07-15:
I think that’s very reasonable.  And I would not worry about fixing the simulation construction set pieces in dev — those are very far from working at this point.  (we should consider removing them completely)

jwnimmer-tri, 2016-07-15:
Thanks for the work here, all.  It sounds like we don't need the ode4 from mathworks, but for the record -- I looked at the license again; it seems restrictive enough to me that I'd want an attorney's blessing before we add it to Drake.

PR #2816 (2016-07-19): Remove ode4
Resolves #2788 . The other file `SpringFlamingoPlant.m` also calls `ode4`. It should be removed also, as mentioned in #2814 .

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2816)
closed 2016-07-19 by hongkai-dai