RobotLocomotion/drake #2797: Mysterious CI Flake - gcc-linux-matlab - fatal error: setoper.h: No such file or directory

liangfok, 2016-07-13:
# The Error

```
05:09:16 make[9]: Entering directory `/home/ubuntu/workspace/experimental/06107e1d/externals/iris/pod-build/iris_project-prefix/src/iris_project-build'
05:09:16 [  2%] Building C object CMakeFilIn file included from /home/ubuntu/workspace/experimental/06107e1d/externals/iris/iris/src/geometry.cpp:5:0:
05:09:16 /home/ubuntu/workspace/experimental/06107e1d/externals/iris/iris/src/iris_cdd.cpp:2:21: fatal error: setoper.h: No such file or directory
05:09:16  #include "setoper.h"
05:09:16                      ^
05:09:16 compilation terminated.
```

# Example Logs

https://drake-jenkins.csail.mit.edu/job/experimental/2658/compiler=gcc,label=linux_matlab/consoleFull

jamiesnape, 2016-07-27:
FYI @mwoehlke-kitware.

RussTedrake, 2016-09-17:
i am now seeing this repeatedly on my local mac clang.  @rdeits -- have you seen this before?

rdeits, 2016-09-19:
Very strange. That header should be included with cddlib, and it should get installed when IRIS builds cddlib. Perhaps this is a similar situation to #1308 in that I am assuming that the cddlib headers will be available after building the library? 

@RussTedrake can you check if you have the file `setoper.h` and if cmake has installed it anywhere?

mwoehlke-kitware, 2016-09-19:
Possibly irrelevant, but if this header comes from an external library (i.e. not iris), it is suspicious that it is `""`-included rather than `<>`-included... it's also a bit suspicious that the header has such a generic name (i.e. doesn't start with `ccd/` or some such).

RussTedrake, 2016-09-19:
The file is there in the source, but not installed

```
drake008% find .. -iname setoper.h
../externals/iris/pod-build/cdd_project-prefix/src/cdd_project/lib-src/setoper.h
../externals/iris/pod-build/cdd_project-prefix/src/cdd_project/lib-src-gmp/setoper.h
```

rdeits, 2017-01-26:
I believe this will be fixed by https://github.com/rdeits/iris-distro/pull/52

PR #4944 (2017-01-30): update IRIS version to remove swig, mex, and pods dependencies
This updates to the latest IRIS master, which uses pybind11 instead of SWIG and which does not build any Matlab mex files or use any of the Pods architecture. 

This *should* preserve IRIS compatibility with Drake, since the Matlab functions still work (they just no longer have mex-ed speedups). However, there's one significant change: Previously, IRIS (like all our other Matlab tools) installed an `addpath_iris.m` file which drake's `checkDependency` would call to mutate the global path and add the IRIS folders. Now, instead, IRIS [... cut]
closed 2017-01-30 by jamiesnape