RobotLocomotion/drake #2798: Mysterious CI Flake - Slave went offline during the build

liangfok, 2016-07-13:
# The Error

```
Slave went offline during the build
05:22:09 ERROR: Connection was broken: java.io.IOException: Unexpected termination of the channel
05:22:09 	at hudson.remoting.SynchronousCommandTransport$ReaderThread.run(SynchronousCommandTransport.java:50)
05:22:09 Caused by: java.io.EOFException
05:22:09 	at java.io.ObjectInputStream$PeekInputStream.readFully(ObjectInputStream.java:2351)
05:22:09 	at java.io.ObjectInputStream$BlockDataInputStream.readShort(ObjectInputStream.java:2820)
05:22:09 	at java.io.ObjectInputStream.readStreamHeader(ObjectInputStream.java:804)
05:22:09 	at java.io.ObjectInputStream.<init>(ObjectInputStream.java:302)
05:22:09 	at hudson.remoting.ObjectInputStreamEx.<init>(ObjectInputStreamEx.java:48)
05:22:09 	at hudson.remoting.AbstractSynchronousByteArrayCommandTransport.read(AbstractSynchronousByteArrayCommandTransport.java:34)
05:22:09 	at hudson.remoting.SynchronousCommandTransport$ReaderThread.run(SynchronousCommandTransport.java:48)
05:22:09 
05:22:09 Build step 'Conditional step (single)' marked build as failure
05:22:09 ERROR: drake-linux-2016-07-01-c4.8xlarge (i-f7dd4d67) is offline; cannot locate Default
```

# Example Logs

### clang-mac
* https://drake-jenkins.csail.mit.edu/job/experimental/2148/compiler=clang,label=mac/consoleFull

## gcc-linux
* https://drake-jenkins.csail.mit.edu/job/experimental/2658/compiler=gcc,label=linux/console

## gcc-linux-matlab
* https://drake-jenkins.csail.mit.edu/job/experimental/2657/compiler=gcc,label=linux_matlab/console

jamiesnape, 2016-07-13:
Network issues.

jamiesnape, 2016-07-15:
I have increased the SSH and TCP timeouts, but really there is nothing to do about this, unfortunately.

closed 2016-07-15 by jamiesnape