RobotLocomotion/drake #2799: lcm-gen failed on bot-lcmgl

RussTedrake, 2016-07-13:
i've been trying to test all of the open PRs on mac, but can't even build master this morning.  I'm currently failing on
```
[ 23%] Performing configure step for 'bot2-lcmgl'
loading initial cache file /Users/russt/drake-distro/build/externals/libbot/bot2-lcmgl-prefix/tmp/bot2-lcmgl-cache-Release.cmake
CMake Error at cmake/lcmtypes.cmake:115 (message):
  lcm-gen failed
Call Stack (most recent call first):
  cmake/lcmtypes.cmake:166 (lcmgen)
  cmake/lcmtypes.cmake:479 (lcmtypes_build_c)
  CMakeLists.txt:8 (lcmtypes_build)

```
not sure yet what changed...

RussTedrake, 2016-07-13:
possibly addressed with #2777

jamiesnape, 2016-07-13:
@jhoare @mwoehlke-kitware Please check whether #2777 addresses this.

jhoare, 2016-07-13:
I don't have a way to test this right now but the main reason why I made #2777 was because when I changed generators things would go bad if I did not go through and delete the ``pod-build`` folders that were created in libbot. 

The command I was running was (from drake-distro): 
```
 $ find . -name pod-build -exec rm -rf {} \;
```

I don't think that's the error you're experiencing, but I can't know for sure.

RussTedrake, 2016-07-14:
that was it.  thanks.

closed 2016-07-14 by RussTedrake