RobotLocomotion/drake #2802: Update RigidBodySystem::GetSensors() API to Not Require Type Dispatch

liangfok, 2016-07-14:
Currently, the only way to obtain the sensors in a `RigidBodySystem` is via `GetSensors()`. This requires down-casting the return values to specific types of sensors, which is against the rules:

https://google.github.io/styleguide/cppguide.html#Run-Time_Type_Information__RTTI_

jwnimmer-tri, 2016-07-14:
FWIW, the cppguide would permit code that called `GetSensors()` and then iterated and checked `sensor->get_name()` to infer the type of the sensor (if it somehow knew what names were what types, e.g., it knew the SDF details), and then downcast to the specific type.  This is because in cppguide _"When the logic of a program guarantees that a given instance of a base class is in fact an instance of a particular derived class, then a dynamic_cast may be used freely on the object. "_

So while, the API may still need changing, but there is at least one valid way to use the current API -- it's not absolutely cppguide-violating, but perhaps it's too difficult to use in a cppguide-compliant manner.

Also note that, if this does get changed, the `RigidBodySystem` API is perhaps not the place to fix it.  If the `RigidBodySensor` class provided methods to interrogate the type of sensor (e.g., `get_sensor_type()`, returning a result from an enumeration), then the `RigidBodySystem` would not have to change at all.

Alternatively, the code that is adding the sensors could retain pointers to the specific types, which means that `GetSensors` would not have to be used anyway.

sammy-tri, 2016-07-14:
I feel like using sensor->get_name() (and get_sensor_type) is skirting awfully close to "Do not hand-implement an RTTI-like workaround. The arguments against RTTI apply just as much to workarounds like class hierarchies with type tags.".

Which isn't to say that I actually object all that strongly...

jwnimmer-tri, 2016-07-14:
When you have a "center of the universe class" like `RigidBodySystem` (which is a scene graph and a diagram, all in one), it's going to be hard to avoid a class hierarchy that provides introspection capabilities.

liangfok, 2016-07-14:
What about adding a whole bunch of `get_foo_sensors()` methods, one for each type of sensor? Poor form?

jwnimmer-tri, 2016-07-14:
What does the calling code want?  If sensors were added programmatically, shouldn't it keep the aliases around itself?  If they came from an SDF/URDF, isn't lookup-by-name already good enough?

liangfok, 2016-07-14:
The calling code simply wants to extract the relevant sensor state and publish it on a ROS topic.

In the current design, the system that takes the sensor state and packages it in a ROS (or LCM) message, does not know which SDF and URDF files were loaded into the `RigidbodySystem`. Thus, it cannot derive the sensor type based on the name unless we establish some kind of sensor-name-ontology.

I'm planning on modifying the SDF / URDF parsers to emit a `ModelSpec` object that is a container of everything needed by `RigidBodySystem` and `RigidbodyTree`. The `ModelSpec` is created by the parsers while they are parsing a SDF or URDF. Thus, the `ModelSpec` will have references to the sensors and could organize them by type. In other words, I could provide a `ModelSpec::get_foo_sensors()` method for each sensor type. Would that be the best way to go?

liangfok, 2016-07-14:
Actually, keeping things in `ModelSpec` will not be a good choice because there will be one `ModelSpec` for each model added to the `RigidBodySystem` and `RigidBodyTree`. The SDF parser can easily emit a vector of multiple `ModelSpec` objects when it parses a single SDF file.

jwnimmer-tri, 2016-07-14:
At this point I'm not even sure what this ticket is about.

liangfok, 2016-07-14:
It's about how to increase the expressiveness of `RigidBodySystem::GetSensors()` to not require RTTI. I suppose introducing a sensor-name ontology that enables the calling code to identify the type of a sensor would be sufficient for this. For example, we can create a rule that says "any sensor whose name contains the string 'lidar' is a `RigidBodyDepthSensor`". Would this be OK given the C++ style guide?

jwnimmer-tri, 2016-12-22:
System 1.0.  Not relevant.

closed 2016-12-22 by jwnimmer-tri