RobotLocomotion/drake #2803: add_mex is not working on my mac

RussTedrake, 2016-07-14:
yet another one.  this happens on a completely clean build on master on my home laptop:

```
[ 21%] Linking CXX shared module /Users/russt/drake-distro/drake/util/barycentricInterpolation.mexmaci64
Undefined symbols for architecture x86_64:
  "_mexErrMsgIdAndTxt", referenced from:
      _mexFunction in barycentricInterpolation.cpp.o
  "_mexPrintf", referenced from:
      _mexFunction in barycentricInterpolation.cpp.o
  "_mxCreateDoubleMatrix_730", referenced from:
      _mexFunction in barycentricInterpolation.cpp.o
  "_mxCreateNumericMatrix_730", referenced from:
      _mexFunction in barycentricInterpolation.cpp.o
  "_mxGetCell_730", referenced from:
      _mexFunction in barycentricInterpolation.cpp.o
  "_mxGetData", referenced from:
      _mexFunction in barycentricInterpolation.cpp.o
  "_mxGetM", referenced from:
      _mexFunction in barycentricInterpolation.cpp.o
  "_mxGetN", referenced from:
      _mexFunction in barycentricInterpolation.cpp.o
  "_mxGetNumberOfElements", referenced from:
      _mexFunction in barycentricInterpolation.cpp.o
  "_mxGetPr", referenced from:
      _mexFunction in barycentricInterpolation.cpp.o
```

because (with VERBOSE=1) it's not actually linking in the matlab libs
```
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++   -Werror=all -Werror=ignored-qualifiers -DGTEST_DONT_DEFINE_FAIL=1 -DGTEST_DONT_DEFINE_SUCCEED=1 -DGTEST_DONT_DEFINE_TEST=1 -bundle -Wl,-headerpad_max_install_names  -o /Users/russt/drake-distro/drake/util/barycentricInterpolation.mexmaci64 CMakeFiles/barycentricInterpolation.dir/barycentricInterpolation.cpp.o  -L/Users/russt/drake-distro/build/drake/lib  -L/Users/russt/drake-distro/build/install/lib
```

if it helps, the only occurence of the mex libraries I could find were:
```
drake008% more CMakeCache.txt| grep -i libmex
Matlab_MEX_LIBRARY:FILEPATH=/Applications/MATLAB_R2015b.app/bin/maci64/libmex.dylib
FIND_PACKAGE_MESSAGE_DETAILS_Matlab:INTERNAL=[/Applications/MATLAB_R2015b.app/extern/include][/Applications/MATLAB_R2015b.app/bin/maci64/libmex.dylib][mexmaci64][/Applications/MATLAB_R2015b.app][cfound components:  MAIN_PROGRAM ][v8.6()]
```

RussTedrake, 2016-07-14:
oddly enough, i do see
```
drake008% grep -R MEX_ * | grep bary
util/CMakeFiles/barycentricInterpolation.dir/flags.make:CXX_FLAGS =  -Werror=all -Werror=ignored-qualifiers -DGTEST_DONT_DEFINE_FAIL=1 -DGTEST_DONT_DEFINE_SUCCEED=1 -DGTEST_DONT_DEFINE_TEST=1 -fPIC   -DMATLAB_MEX_FILE -I"/Applications/MATLAB_R2015b.app/extern/include" -I"/Applications/MATLAB_R2015b.app/simulink/include" -fno-common -arch x86_64 -mmacosx-version-min=10.11 -fexceptions -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk -DMATLAB_MEX_FILE -DMATLAB_MEX_FILE -DMATLAB_MEX_FILE  -std=gnu++14
```

how am i supposed to build just drake with the new out-of-source build scripts?
if i run build in drake-distro/build/drake, then it appears to do the right thing, but fails with the error above.  if i cd into drake-distro/drake and run make, then i run into different issues:
```
drake008% make
Configuring with CMAKE_FLAGS: -DCMAKE_BUILD_TYPE=Release
-- CMAKE_INSTALL_PREFIX=/Users/russt/drake-distro/build
-- Found matlab
CMake Warning at CMakeLists.txt:157 (message):
  MATLAB JVM version (1.7) does not match installed Java SDK version (1.8)

CMake Error at /usr/local/Cellar/cmake/3.5.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find GTest (missing: GTEST_LIBRARY GTEST_INCLUDE_DIR
  GTEST_MAIN_LIBRARY)
Call Stack (most recent call first):
  /usr/local/Cellar/cmake/3.5.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
  /usr/local/Cellar/cmake/3.5.2/share/cmake/Modules/FindGTest.cmake:205 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:224 (find_package)

-- Configuring incomplete, errors occurred!
See also "/Users/russt/drake-distro/drake/pod-build/CMakeFiles/CMakeOutput.log".
make: *** [configure] Error 1
```

RussTedrake, 2016-07-14:
ok.  figured out this problem (but have not resolved it).  basically, the default BUILD_TYPE is no longer being set by the cmake scripts
```
cmake ../../drake -DCMAKE_BUILD_TYPE:STRING=Release
```
works (generates make files which will compile successfully), but 
```
cmake ../../drake
```
does not.  because around line 344 in cmake/mex.cmake is setting the library flags for each of the four configurations i know about explicitly.  

all of these build changes have absolutely killed my productivity.

RussTedrake, 2016-07-14:
btw - having the from_source instructions recommend that people build everything in debug mode, including the externals, is a major flaw.  as soon as you try to run any sizable examples (e.g. simulating atlas) in debug mode, you'll see that all of the Eigen runtime checks bring the performance make everything painfully slow.  I believe the from_source instructions should get people building in release mode, and the developer instructions should tell people how to build in debug mode.  +@david-german-tri

wxmerkt, 2016-07-14:
Just wondering, is ``RelWithDebInfo`` a potential solution to this?

jhoare, 2016-07-14:
If you want to set a default build type, we could add the following to the top-level CMakeLists.txt

```
if(NOT CMAKE_BUILD_TYPE)
  set(CMAKE_BUILD_TYPE "Release" CACHE STRING 
      "Choose the type of build, options are: Debug Release RelWithDebInfo MinSizeRel."
      FORCE)
endif(NOT CMAKE_BUILD_TYPE)
```

We could set this to ``Release`` or ``ReleaseWithDebInfo`` and then most users would not need to set it when building, and developers could override it with ``Debug`` if they desire.

jamiesnape, 2016-07-14:
@RussTedrake Will @jhoare's solution work for you? I would set it to `Release` since that is what we use for the majority of our testing.

RussTedrake, 2016-07-14:
Yes.  That is the logic that we had before.

david-german-tri, 2016-07-14:
I can repro this.  Testing now whether #2807 fixes it.

david-german-tri, 2016-07-14:
Confirmed that #2807 is a fix on my Mac.

PR #2807 (2016-07-15): Set a CMake Default build type when one is not defined.
  Fixes #2803 

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2807)
commit 5d120994 (2016-07-15): Set a CMake Default build type when one is not defined.

  Fixes #2803
closed 2016-07-15 by RussTedrake