RobotLocomotion/drake #2808: Document the Locations of the "Build" and "Install" Directories

liangfok, 2016-07-14:
Prior to #2706, they were `drake-distro/pod-build` and `drake-distro/build`, respectively.

Post #2706, they are `drake-distro/build` and `drake-distro/build/install`, respectively.

david-german-tri, 2016-07-14:
To be clear, the statements in the original post are for `make` builds.

liangfok, 2016-07-14:
What are the build and install directories for ninja-based builds?

david-german-tri, 2016-07-14:
Pre-#2706, we recommended `drake-build` (and thus `drake-build/install`).

david-german-tri, 2016-07-14:
As part of this issue, we should also check for hard-coded paths thoughout Drake.

RussTedrake, 2016-07-15:
yes -- searching for 'pod-build' in drake reveals a number of tests and pieces of documentation that will be broken.

david-german-tri, 2016-07-15:
All the MATLAB references to `pod-build` are now cleaned up in #2599.

david-german-tri, 2016-07-18:
This isn't totally closed because #2599 hasn't merged yet.

PR #2824 (2016-07-18): Added details about locations of build and install directories. 
Closes #2808.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2824)
PR #2599 (2016-07-20): Update MATLAB code to support out-of-source builds.
Add search logic to the hand-coded addpath_drake.m for the common out-of-source build cases. This allows addpath_drake to work correctly from the source-tree side.  Update the generated addpath_drake wrapper to use the `CMAKE_INSTALL_PREFIX`.  Add regression test coverage.  Generalize other uses of the build directory throughout the MATLAB code base.

Resolves #2808.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2599)
closed 2016-07-20 by david-german-tri