RobotLocomotion/drake #2811: Rename "test" package to be something more specific

liangfok, 2016-07-14:
The "test" package is defined here:

https://github.com/RobotLocomotion/drake/blob/af7705b950469e7d477476e2c3b0fe56a4f1b3e4/drake/systems/plants/test/package.xml

Since `xmlUtil.cpp` enforces that there be only one package of a particular name through Drake, we should re-name it something more specific.

jwnimmer-tri, 2017-03-11:
@liangfok I think this file is gone now?

liangfok, 2017-04-12:
Yes, I believe the "test" package is gone now. The following was obtained using fb409c870e6bd7d76f27b6cb7a7c7e5f4ae64798.

```
$ cd drake-distro
$ find . -name "package.xml" | grep test
./drake/multibody/parsers/test/package_map_test/package_map_test_packages/package_map_test_package_a/package.xml
./drake/multibody/parsers/test/package_map_test/package_map_test_packages/package_map_test_package_b/package.xml
./drake/multibody/parsers/test/package_map_test/package_map_test_packages/package_map_test_package_set/package_map_test_package_c/package.xml
./drake/multibody/parsers/test/package_map_test/package_map_test_packages/package_map_test_package_set/package_map_test_package_d/package.xml
./ros/drake_ros_common_test/package.xml
```

closed 2017-04-12 by liangfok