RobotLocomotion/drake #2813: probably lost lcm backwards compatibility?

RussTedrake, 2016-07-15:
it occurred to me that if the new build logic can no longer find and use lcm from pkg-config, then we have almost certainly broken the open-humanoids build again.  At the very least, OH will have to upgrade to the latest LCM to make things work.  (Is there a minimum version number?)

@wxmerkt -- perhaps you can confirm?

wxmerkt, 2016-07-15:
Thanks for the heads-up, Russ. 
OH recommends v1.3.0, Drake now requires a LCM version newer than v1.3.1, and none has been released yet with the new build system - cf. [releases](https://github.com/lcm-proj/lcm/releases). Perhaps a new LCM version v1.4.0 should be released and tagged with the new build system and Drake's requirements set to >1.4.0.

Upgrading LCM most likely wouldn't be an issue for OH given that that LCM >1.3.1 maintains pkg-config compatibility (cf. [here](https://github.com/lcm-proj/lcm/pull/75/files#diff-af3b638bc2a3e6c650974192a53c7291R84)); yet we would have to do it on every machine manually given that it's a system-installed dependency. 
We haven't updated Drake in over four months (and are currently using a slightly-diverged fork that Siyuan/Lucas worked on) given that the number of recent API changes prompt a rather large software engineering effort to keep up with.

RussTedrake, 2016-07-15:
Thanks.  I know that you guys haven't updated drake recently.  @gizatt has been working on a fork which is trying to keep up with drake and will be able to merge that back in soon.  It won't be long now til we have a new c++ val sim that will be a big incentive for you to update.  :)

So it seems that as long as we can support lcm being installed to a system path, instead of being a part of the superbuild, then we should be ok (just have to force folks to upgrade).  I presume that this will work... We just have to make sure that FindLCM logic still works, but also that it hasn't been tested.

It would possibly also be good to increment a version number in lcm and set the min version in drake.  Or specifically to make sure that if drake does not find lcm (because a pkg-config only version is installed) that it prints a useful error stating the min rev.

mwoehlke-kitware, 2016-07-18:
> We just have to make sure that FindLCM logic still works, but also that it hasn't been tested.

Modern CMake usage via `find_package(lcm)` relies on finding the CMake package configuration file (`lcmConfig.cmake`) in a default search path. For a system-provided LCM, this will Just Work™, as long as that file is installed...

(...to the correct subdirectory of the install prefix, along with the necessary supporting files. Which shouldn't be an issue as long as the distro doesn't botch packaging; after all, it's already being done successfully in the drake superbuild. The only variance is using drake's install prefix vs. the system install prefix, and there's no reason to believe that would break anything.)

RussTedrake, 2016-07-18:
that's what i would hope.  to test, i would recommend
 - downloading and installing LCM to it's own directory, following the
instructions on the LCM website.  (e.g. run make install)
 - set WITH_LCM to OFF in drake-distro
 - confirm that drake still finds LCM.

mwoehlke-kitware, 2016-07-20:
I played around with this a bit, installing a stand-alone LCM into the `install` subdirectory of its build directory (so, a very non-standard install prefix). This obviously requires setting `CMAKE_PREFIX_PATH` if anything is to have any hope of finding LCM. I did run into some issues:

- `CMAKE_PREFIX_PATH` should be propagated to externals. This is fixed on my `superbuild-refactor-part-2` branch (which needs to be rebased and is waiting on #2820).
- Users of the LCM types need to include LCM's include directory, but aren't doing so. These are technically broken already and are only working due to "luck". (Missing include directories is actually a pretty common error in C++ software, because it doesn't show up until the only copy of a dependency is in a strange and unique location.) At some point I'm going to make a PR for LCM to add a `lcm-coretypes` target that such users can link to get the includes without having to actually link to the full LCM library. We also ought to be creating `INTERFACE` targets to represent the C++ lcmtypes "libraries".

None of this should be a problem for installing into a system path (as e.g. a distro package would do), however.

RussTedrake, 2016-07-21:
Thanks.  I believe that the include directory used to come through the pkg-config scripts, and that it was working correctly?

mwoehlke-kitware, 2016-07-21:
> I believe that the include directory used to come through the pkg-config scripts, and that it was working correctly?

I'm not entirely sure of that... users of *the lcm types "libraries"¹* (drake's, bot2_core's) need the LCM includes. If it was working, they would need to `include_directories(${LCM_INCLUDE_DIRS})` (which is anyway discouraged in modern CMake in favor of `target_include_directories`). Such consumers may or may not be directly using LCM, and so may or may not have called `pods_use_pkg_config_packages(lcm)`. Even if so, we're trying to reduce/remove use of pkg-config.

Obviously, things are working at the moment, because the LCM headers are in a path that is included for other reasons. So this issue will only manifest if someone is using their own version of LCM that is *not* provided by the system *and* is in a non-default install prefix (i.e. other than `/usr/local`).

I'd also be surprised if LCM is the only package that's in this boat. I'd certainly agree we should try to clean up this sort of thing, but it's something of a hypothetical rather than genuine problem. Typically, this sort of issue is latent until someone actually tries to build using an esoteric configuration.

(Moreover, it's only a problem for libraries using the LCM C++ types "libraries" but *not* using LCM itself. Which... exist, clearly, though I'm not entirely sure why.)

Again, I want to be clear: if we end up encouraging people to use a *distro-provided* (or otherwise installed into `/usr`) LCM, I don't believe the issue is going to manifest.

(¹ LCM C++ types "libraries" are header-only.)

RussTedrake, 2016-07-21:
I just want to make sure that the movement away from pods and pkg config is not introducing new issues.  The cmake pods_pkg_config_* macros/functions would have called the cflags, if used properly.

david-german-tri, 2016-12-13:
It sounds to me like there are no action items left in this issue (and it hasn't been a pain point in the past five months, to the best of my knowledge), so I'll close it.

closed 2016-12-13 by david-german-tri