RobotLocomotion/drake #2817: CTest is not respecting the return values that are explicitly set in the CI driver script

jamiesnape, 2016-07-15:
Consolidates #2342 and #2791.

I consider it high priority, but I do not think others agree, so @david-german-tri can you assign a priority?

david-german-tri, 2016-07-15:
Could you elaborate the common cause of failure at work here?  Perhaps with code links to the return values that are being ignored?

jamiesnape, 2016-07-15:
https://github.com/RobotLocomotion/drake-ci/blob/master/ctest_driver_script.cmake#L1301

jamiesnape, 2016-07-15:
And the implicit return 0 at the end of that script. I think we went over this in detail F2F on Wednesday.

david-german-tri, 2016-07-19:
Ah, right, I remember now - you were saying CTest does some kind of deeper inspection of the intermediate test results.  I agree this is high priority, but you gave me the impression it was low feasibility, because upstream changes to CTest would be required?

jamiesnape, 2016-07-19:
Since we maintain CTest, it is feasible. It just may take a little time.

david-german-tri, 2016-07-19:
OK, sounds good!  Have at it.

jamiesnape, 2016-10-05:
Fixed in upcoming CMake 3.7.

billhoffman, 2016-10-06:
Leaving this open, when we move to 3.7 for CI we can do the switch, it will mean changes to the ctest scripts.

david-german-tri, 2016-12-15:
from f2f conversation with @jamiesnape, this is fixed and live in CI.

jamiesnape, 2016-12-20:
Unfortunately.

jamiesnape, 2016-12-20:
No longer live.

jamiesnape, 2016-12-20:
@billhoffman `CMAKE_CAPTURE_ERROR` appears to be giving different results to `RETURN_VALUE` for at least `ctest_build()` and `ctest_test()`.

jamiesnape, 2016-12-20:
Polar-opposite results in some cases.

EricCousineau-TRI, 2018-02-08:
@jamiesnape Is this still relevant?

closed 2018-02-08 by jamiesnape