RobotLocomotion/drake #2822: Continuous mac builds are failing tests

jwnimmer-tri, 2016-07-15:
- Full the list of status https://drake-jenkins.csail.mit.edu/view/Continuous/
- Example first failure https://drake-jenkins.csail.mit.edu/view/Continuous/job/mac-clang-continuous-release/360/
- Example CDash output https://drake-cdash.csail.mit.edu/viewTest.php?onlyfailed&buildid=51370
- Failing tests are runPendulumDynamics and runQuadrotorDynamics, both had a segfault.
- These two tests on mac started always failing after #2697 was merged, and never before.

jwnimmer-tri, 2016-07-15:
Perhaps #2818 and #2819 were attempts to diagnose and revert this; unclear why #2697 wasn't also tested as a revert.

jwnimmer-tri, 2016-07-15:
Aha; #2697 was just unlucky as the first merge to happen after the OS X VM image was rebuilt.  I guess something in the image is incompatible with those tests.

liangfok, 2016-07-16:
I've seen that "LCM self test failed" error on someone else's computer. Most likely a firewall issue.

jamiesnape, 2016-07-18:
Only just saw this. I will check the firewall settings. I need to allow UDP from something like 224.0.0/4?

jamiesnape, 2016-07-18:
Should we be setting `LCM_DBG` at all on the builds? Not sure how useful or verbose it is.

jamiesnape, 2016-07-18:
Routing problem. Fixed.
* https://drake-jenkins.csail.mit.edu/view/OS%20X/job/mac-clang-continuous-debug/394/
* https://drake-jenkins.csail.mit.edu/view/OS%20X/job/mac-clang-continuous-release/368/
* https://drake-jenkins.csail.mit.edu/view/OS%20X/job/mac-clang-ninja-continuous-debug/173/
* https://drake-jenkins.csail.mit.edu/view/OS%20X/job/mac-clang-ninja-continuous-release/174/

closed 2016-07-18 by jamiesnape

liangfok, 2016-07-19:
> I will check the firewall settings. I need to allow UDP from something like 224.0.0/4?

I'm not sure what the subnet mask should be, but the IP address of `224.0.0.0` appears to be correct based on this: https://github.com/RobotLocomotion/lcm/blob/master/lcm/udpm_util.c#L322

> Should we be setting LCM_DBG at all on the builds? Not sure how useful or verbose it is.

I think the default of `DBG_ERROR` should be sufficient. https://github.com/RobotLocomotion/lcm/blob/master/lcm/dbg.h#L35