RobotLocomotion/drake #2827: Incremental submodule update not working after lcm repository switched

jwnimmer-tri, 2016-07-18:
Reproduction steps:
```
git clone 'https://jwnimmer-tri@github.com/jwnimmer-tri/drake.git' drake-distro
cd drake-distro
git remote add upstream 'https://jwnimmer-tri@github.com/RobotLocomotion/drake.git'
git remote set-url --push upstream no_push
git reset --hard 8535fc31ec624e632b05c6212a7134947a59a28f   # Just before #2809.
mkdir -p build/debug
cd build/debug
env PATH=/usr/lib/ccache:$PATH CC=gcc-4.9 CXX=g++-4.9 \
  cmake ../../ -G Ninja -DCMAKE_BUILD_TYPE:STRING=Debug
env PATH=/usr/lib/ccache:$PATH CC=gcc-4.9 CXX=g++-4.9 ninja -j1 -v
git fetch upstream && git merge --ff-only upstream/master  # Takes on #2809.
env PATH=/usr/lib/ccache:$PATH CC=gcc-4.9 CXX=g++-4.9 ninja -j1 -v
```

The last build line (after merging 2809, then re-rerunning ninja), yields:
```
[27/88] cd /home/jwnimmer/jwnimmer-tri/drake-distro/externals/lcm && /home/jwnimmer/jwnimmer-tri/cmake-3.5.2-Linux-x86_64/bin/cmake -E chdir /home/jwnimmer/jwnimmer-tri/drake-distro /usr/bin/git submodule update --init --recursive -- externals/lcm && /home/jwnimmer/jwnimmer-tri/cmake-3.5.2-Linux-x86_64/bin/cmake -E touch /home/jwnimmer/jwnimmer-tri/drake-distro/build/debug/lcm-prefix/src/lcm-stamp/lcm-update
FAILED: cd /home/jwnimmer/jwnimmer-tri/drake-distro/externals/lcm && /home/jwnimmer/jwnimmer-tri/cmake-3.5.2-Linux-x86_64/bin/cmake -E chdir /home/jwnimmer/jwnimmer-tri/drake-distro /usr/bin/git submodule update --init --recursive -- externals/lcm && /home/jwnimmer/jwnimmer-tri/cmake-3.5.2-Linux-x86_64/bin/cmake -E touch /home/jwnimmer/jwnimmer-tri/drake-distro/build/debug/lcm-prefix/src/lcm-stamp/lcm-update
fatal: reference is not a tree: 9c9e22cf23663ef9cd0e5ec83932ff7f70cd7b18
Unable to checkout '9c9e22cf23663ef9cd0e5ec83932ff7f70cd7b18' in submodule path 'externals/lcm'
ninja: build stopped: subcommand failed.
```

jwnimmer-tri, 2016-07-18:
Tagging #2757 as the broader discussion for related issues.

jwnimmer-tri, 2016-07-18:
@bradking What am I doing wrong here?

jwnimmer-tri, 2016-07-18:
/CC @david-german-tri

jwnimmer-tri, 2016-07-18:
I guess the work-around of nuking `drake-distro` and cloning again is plausible, so I'll drop this down to medium.

bradking, 2016-07-18:
The problem is that the `git submodule sync` step only runs when CMake re-runs, and in this particular case nothing caused CMake to re-run.  I'll look at making a build step that does it too.

bradking, 2016-07-18:
@jwnimmer-tri all you need to do is run `cmake .` at the top of your build tree to move forward for now.

PR #2828 (2016-07-19): cmake: Sync submodule URL changes during build
When `AUTO_UPDATE_EXTERNALS` is `ON` we are responsible for keeping external source trees up to date.  Previously we achieved this by running `git submodule sync` during CMake configuration to honor changes to the submodule URLs.  However, it is possible for an incremental update to re-build without re-running CMake.  Therefore we need to do the sync at the start of the build instead.

Fixes #2827.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2828)
commit e374fe7e (2016-07-19): cmake: Sync submodule URL changes during build

When `AUTO_UPDATE_EXTERNALS` is `ON` we are responsible for keeping
external source trees up to date.  Previously we achieved this by
running `git submodule sync` during CMake configuration to honor changes
to the submodule URLs.  However, it is possible for an incremental
update to re-build without re-running CMake.  Therefore we need to do
the sync at the start of the build instead.

We need to add dependencies very carefully to make sure this works with
the CMake Ninja generator.  Any target to which CMake may propagate the
download step custo [... cut]
closed 2016-07-19 by jwnimmer-tri