RobotLocomotion/drake #2830: On Ubuntu 14.04, can CMake automatically default to `CXX=g++4.9` and etc?

jwnimmer-tri, 2016-07-18:
On Ubuntu 14.04, can Drake's superbuild CMake automatically default to `CXX=g++4.9` and related (`CC`, fortran, etc.) if they are not explicitly set by the environment (and/or if the as-invoked default is too old; and/or are not already in the cache)?

The documentation and instructions to set up `CXX` are error-prone and annoying to users.  I've had two different teams try to use `update-alternatives` to make it move convenient, with the obvious disastrous results.

I tried googling for this a month ago, seemed like it was difficult, and so gave up.  But perhaps Kitware has a solution at-hand.

jamiesnape, 2016-07-19:
No. The compiler needs to be setup beforehand. The best you can do is have some shell script that developers source to set up their environment.

liangfok, 2016-07-19:
+1 for providing a shell script that sets the appropriate environment variables.

david-german-tri, 2016-07-19:
Writing a One True Shell Script is on my todo list for this quarter (but for 16.04).

jamiesnape, 2016-07-19:
My thinking that this would be one simply setting environment variables, separate from one that installs external dependencies from package managers. You would source this before building in a new shell.

liangfok, 2016-07-19:
+1 for separating the script that sets environment variables from the one that installs dependencies.

david-german-tri, 2016-09-08:
We have stable instructions for 14.04, and 16.04 is obviously the future, so I'm going to close this as will-not-fix.

closed 2016-09-08 by david-german-tri