RobotLocomotion/drake #2836: Design System2 semantics for "publishers": systems that have side effects

david-german-tri, 2016-07-19:
@liangfok's #2821 is the first System2 system where the `EvalOutput` call has a side effect. namely, sending an LCM message.  Under the current API, this side effect is invisible to the solver.  Is that really what we want, or should the solver have some kind of explicit handle for doing things that cause side effects?

RussTedrake, 2016-07-19:
I like the question, but i don't see this as a significant side effect; it can be as innocusous as a pirnt statement (would you consider that a side effect?)

liangfok, 2016-07-19:
I suppose the challenging issue is what happens if the solver decides to step back in time? Is the solver also responsible for "un-doing" previously sent LCM messages? If so, it will need to know when LCM messages are transmitted as a side effect of `EvalOutput()`.

david-german-tri, 2016-07-19:
Yeah, I think a print statement has the same set of problems.  If `EvalOutput` is the only place you can put your print statements, they will execute or not according to whether the solver needs your output - perhaps at erratic intervals, perhaps not even monotonically in sim time.  Maybe that's OK, but it seems weird to me.

Strawman alternative: you could imagine a `void Publish(t, x, u) const` method with a fixed sample rate.  The solver must call `Publish` at exactly its sample rate, but since publication doesn't affect outputs or state derivatives it is compatible with a continuous Diagram.  Ditto for an event or witness driven Publish.

sherm1, 2016-07-19:
I see I'm behind on this discussion. Here is my comment from #2821:
> We need something better than a side effect of EvalOutput as a way to get a System to perform work that is invisible to the System diagram abstractions. In general a System has multiple output ports with different sampling rates -- in this case there are zero output ports so I wouldn't expect a solver to call EvalOutputPort at all. A System block itself has a sampling rate also and we should probably be triggering its side-effect activity off the block sample rather than off a (non-existent) output port sample.

david-german-tri, 2016-07-19:
@sherm1 :+1:, with the caveat/elaboration that even purely continuous systems can have a "sample rate" for the purposes of side effects.

jwnimmer-tri, 2016-07-19:
But do you want to print / publish every time the value changes, or at a fixed clock?  Shouldn't both of those alternatives be supported?

sherm1, 2016-07-19:
@david-german-tri, yes that's what I was thinking too -- continuous counts as a sample rate.

> I suppose the challenging issue is what happens if the solver decides to step back in time? 

That is definitely a problem with side effects that are triggered off outputs because internal outputs must be executed during trial steps in order to produce tentative results. However it would not be a problem with a suitably-designed block-sample method (let's call it `Sample()` here) because that method would be invoked only at non-retractable sampling intervals (either at a specific sample time for discrete System or at new step start for a continuous System).

david-german-tri, 2016-07-19:
@jwnimmer-tri I'm not really sure what "every time the value changes" means for a system that has continuously-variable states and outputs.  The only options I can comprehend are (a) a fixed clock, (b) an event/witness function, or (c) the discretization the solver happens to use, which AIUI comes with no guarantees.  (I imagine e.g. annealing back and forth to zero in on a point where some state variable derivative changes abruptly, but @sherm1 can doubtless offer a better intuition.)

sherm1, 2016-07-19:
> But do you want to print / publish every time the value changes, or at a fixed clock? Shouldn't both of those alternatives be supported?

Yes. Assuming we implement the full set of sampling options, arrival of an interesting message can trigger an event that causes an event-driven sample to occur. An external user action could also trigger an event like that. But internally no "surprise" value changes can occur since all the computations occur with defined sampling specifications.

david-german-tri, 2016-09-08:
@sherm1's approach from #3202 seems good to me.  Closing.

closed 2016-09-08 by david-german-tri