RobotLocomotion/drake #2839: MATLAB on OS X tests failing due segmentation violation

jamiesnape, 2016-07-19:
For example:
* https://drake-cdash.csail.mit.edu/viewTest.php?onlyfailed&buildid=53278
* https://drake-cdash.csail.mit.edu/viewTest.php?onlyfailed&buildid=53338

FYI @RussTedrake.

david-german-tri, 2016-07-19:
Do we know with reasonable certainty that this is due to Java 8, and not due to a PR that landed during the Mac CI hardware outage?

jamiesnape, 2016-07-19:
It could be either, but I would lean toward it being a Java 7/8 issue.

david-german-tri, 2016-07-19:
In that case, we should bisect it.  I've kicked off an experimental build at 41d2dc2f507f36f14d5296ec6d4e440c6eb7132b - it will be #4 here once it starts up:

https://drake-jenkins.csail.mit.edu/view/Experimental/job/mac-clang-ninja-experimental-matlab/#

jamiesnape, 2016-07-19:
https://drake-jenkins.csail.mit.edu/view/Experimental/job/mac-clang-ninja-experimental-matlab/6/

RussTedrake, 2016-07-20:
this is the known issue with the instantaneousQP controller.  see #2165.  those tests should have been disabled.

jamiesnape, 2016-07-27:
FYI, we also have some tests segfaulting in #2916. Once #2947 is fixed, we can evaluate whether to close this ticket.

RussTedrake, 2016-09-08:
@jamiesnape -- please close if/when those tests are disabled.

david-german-tri, 2016-09-08:
My memory is that we disabled these, and mac-clang-ninja-nightly-matlab-{experimental, release} is blue in CI.

closed 2016-09-08 by david-german-tri