RobotLocomotion/drake #2840: Possible alternative API for multiple "glass box" instantiations of Systems

sherm1, 2016-07-19:
Continuing a discussion from last month: I've been attempting to work out a simple API that would support multiple instantiations of complicated Systems (like RigidBodySystem) with non-templatized methods allowing ordinary runtime polymorphism (virtuals), and without the user having to carry around all the instantiations. I would also like to be able to define `MySystem` once, and produce arbitrary instantiations like `MySystem<double>`, `MySystem<AutoDiffScalar<...>>`, `MySystem<FunctionalForm>`, without having to list those possibilities in `MySystem`'s declaration; no code generators required!

I managed to get the following code to work -- what do you think of this API? The basic idea is to have a single "fundamental" system instantiated as `MySystem<double>`. That system is used for simulation, along with a `Context<double>`. Other "alternate" instantiations are created automatically from the fundamental system (so they are identical by construction) If at any point you want access to a different instantiation, you just ask the fundamental system for it, initialize a matching Context from the fundamental Context, and then call whatever methods you like on the alternate system.

```c++
int main() {
  // Fill in the "MultibodyTree" first.
  auto mb_tree = make_unique<MBTree<double>>();

  mb_tree->AddJoint(make_unique<PinJoint<double>>(1.1));
  mb_tree->AddJoint(make_unique<SliderJoint<double>>(2.2));
  mb_tree->AddJoint(make_unique<PinJoint<double>>(3.3));
  mb_tree->AddJoint(make_unique<SliderJoint<double>>(4.4));

  // Create the fundamental MBSystem (that is, type double).
  MBSystem<double> sys(std::move(mb_tree));

  // Create some alternate instantiations of MBSystem (kept within the
  // fundamental system). This could be done in a `.cc` file somewhere so
  // that we don't have to expose the code to user's compilation units.
  MBSystem<complex<double>>::AddAlternate(sys);
  MBSystem<float>::AddAlternate(sys);

  cout << "num alternates=" << sys.get_num_alternates() << endl;

  // Stuff happens using the fundamental system, then at some point we decide
  // we want one of the alternate instantiations.

  // These are very fast. 
  const auto& csys = sys.get_alternate<complex<double>>();
  const auto& fsys = sys.get_alternate<float>();

  cout << "my type=" << sys.type() << endl;
  cout << "csys type=" << csys.type() << endl;
  cout << "fsys type=" << fsys.type() << endl;

  // Dig out the matching instantiations of the multibody tree.
  const auto& dtree = sys.get_mb_tree();   // <double> (fundamental)
  const auto& ftree = fsys.get_mb_tree();  // <float> (not useful)

  // Using the fundamental system, calculate derivative df analytically.
  Context<double> cd{0.5};  // set x=0.5 (set up context for fundamental)

  const auto& dpin0 = dtree.GetJoint<PinJoint>(0);
  double f = dpin0.PinFunc(cd);
  double df = dpin0.DPinFuncDx(cd);  // analytical derivative

  // Instead, use the same joint of the complex alternate to calculate the
  // derivative using a complex step derivative (equivalent to autodiff).
  const auto& ctree = csys.get_mb_tree();
  const auto& cpin0 = ctree.GetJoint<PinJoint>(0);

  Context<complex<double>> cc(cd); // clone context for this alternate.
  cc.x += complex<double>(0, 1e-20);  // complex step derivative
  double cdf = cpin0.PinFunc(cc).imag() / 1e-20;

  printf("  f(x)=%.16g;\n df(x)=%.16g (analytical)\ncdf(x)=%.16g (autodiff)\n",
         f, df, cdf);
}
```

Here is the output from the above code:
```
num alternates=2
my type=double
csys type=class std::complex<double>
fsys type=float
  f(x)=0.8775825618903728;
 df(x)=-0.479425538604203 (analytical)
cdf(x)=-0.4794255386042029 (autodiff)
```

That code differentiates the non-templated function `PinFunc()` using the alternate, but only requires users to keep track of the fundamental system. It would also be possible to have a joint (or other element) locate its own doppelganger in the alternate, although I don't think that's necessary.

Russ has seen this code (he's here in Palo Alto today!) and responded positively.

/cc @david-german-tri, @jwnimmer-tri, @amcastro-tri, @RussTedrake, @tkoolen

sherm1, 2016-07-19:
Here is what the `MySystem` implementation looks like:
```c++
// This is a concrete System class, modeling MultibodySystem.
template <typename T>
class MBSystem : public System<MBSystem, T> {  // CRTP!
  using Super = System<::MBSystem, T>;

 public:
  // Construct and take over ownership of the given MBTree.
  MBSystem(unique_ptr<MBTree<T>> tree) : tree_(std::move(tree)) {}

  // Construct an alternate MySystem<T> with same structure as the given one.
  // This is also the copy constructor for the T=double fundamental.
  explicit MBSystem(const MBSystem<double>& fundamental)
      : Super(fundamental), tree_(new MBTree<T>(*fundamental.tree_)) {}

  // For demo purposes, report the name of the scalar type here.
  const char* type() const { return typeid(T).name(); }

  // Return interesting concrete System-specific stuff.
  const MBTree<T>& get_mb_tree() const { return *tree_; }

 private:
  // Let all other instantiations of this class be friends with this one.
  template <typename TT>
  friend class MBSystem;

  unique_ptr<MBTree<T>> tree_;
};
```

For the actual code, see [this gist](https://gist.github.com/sherm1/fab75e7fdb41a6cedecbcfea36c87cee).

amcastro-tri, 2016-07-19:
The `AlternateInstantiation` idea is great. I like how you solved it with `std::type_index`. 
Now, how would we setup a problem with variable parameters? I mean, in that case the context couldn't just be cloned because the auto-diff version should have the parameters in the cache. Maybe it could be cloned and then we could add the remaining (parameters) entries to the cache?

sherm1, 2016-07-20:
Parameters should be treated exactly like Inputs (in my view they are essentially the same thing except for  when they get updated). Our current strategy for Inputs is that cloning a Context brings in the current values for the inputs, so that the Context is complete. That's how Parameters should work too. So the cloned Context would have values for all the parameters; you would then be able to set the AutoDiff values for the parameters if you want derivatives with respect to parameters.

amcastro-tri, 2016-07-20:
Maybe an example would help. Still not clear to me how the API for that would look like. For instance, say I want to be able to differentiate with respect to `PinJoint::parameter_` of the third joint. How would I set up the problem so that `PinJoint::parameter_` ends up in the context? I imagine that in the current example this parameter is not in the context since it is not a variable of the problem.

bradking, 2016-08-05:
> ```
> template <template <typename> class MySys, typename T> class System
> ...
> /** For a System instantiated with the fundamental scalar type `double` this
> class contains a registry of the other available instantiations of the
> same System. For any other type this struct is empty. **/
> template <typename T>
> class AlternateInstantiations
> ```

In principle no specific `<T>` (e.g. `<double>`) needs to be specially defined as the canonical version.  It may be possible to define a system through any type and register alternates through a common map shared by all associated instances of any type.  This may mean something like

```c++
template <template <typename> class MySys> class System: public InstantiatedSystem
```

as an intermediate derivation level.

sherm1, 2016-08-05:
@bradking - thanks. I did start with a generic registry rather than favoring a specific instantiation. But then I thought that everyone would at least have a `System<double>` object around for computation so I might as well use that rather than add the extra derivation layer and generic interface. I was trying to get close to the Matlab interface that Russ showed me where people work with objects from the "fundamental" (`double`) instantiation (like a particular Joint) and then decide somewhere deep in the code that they need to get a derivative associated with one of those objects. If that is a typical usage pattern then user code will have easy access to the fundamental System already so it makes a good place to keep the alternates.

david-german-tri, 2016-10-31:
This is very slick!  I added a question about the copy constructor to the gist.  With that tweak, I agree this achieves two benefits that no other approach does:

- Auto-transmogrification of Diagrams.
- No requirement to carry around the various instantiations.

My question is: do those benefits justify the complexity, in particular the CRTP?  You could get the exact same transmogrifying behavior in `MBSystem` without touching `system.h`.  You'd lose `AddAlternate/get_alternate` as generic hooks on Systems, but do we really need them?

david-german-tri, 2016-10-31:
Also, we can add framework-level transmogrification on a closed set of types sans CRTP, which might be a compromise worth considering.

sherm1, 2016-10-31:
Since gist comments don't send any notifications, I'll quote David here:
> Instead of a copy constructor on MBSystem, could we have a virtual method System<T>::TransmogrifyFrom(System<double>)? Then Diagrams can Transmogrify generically.

Yeah, that sounds like a better idea (except maybe for the word "Transmogrify"!). `CloneFrom(System<double>)` seems a little less terrifying.

I'm also not sure whether we need the `get_alternate()` layer but it's nice to know it can be done if we want to.

david-german-tri, 2016-10-31:
OK, cool.  I'm working on a proof-of-concept PR for a closed set of types - specifically, `AutoDiffXd`.  It is much less code than this, but I don't think it closes the door on the more general approach if we need it.

amcastro-tri, 2016-11-06:
This is so great! I might start using the pattern proposed in #3967 very soon with RBT since when templating it on `<T>` I need to be able to get an `RBT<AutoDiffScalar>` for the auto-diff unit tests.

Also I think it is worth taking a look at the Matlab code to see what solutions were proposed there already. Those solutions go very well along what you guys are proposing! (though I think the proposals here are much refined). 
For instance take a look a the code creating the mex pointer for `RigidBodyManipulator` (Matlab's RBT) [here](https://github.com/RobotLocomotion/drake/blob/master/drake/matlab/systems/plants/%40RigidBodyManipulator/RigidBodyManipulator.m#L2148). Notice how there are the methods `createKinematicsCachemex` and [`createKinematicsCacheAutoDiffmex`](https://github.com/RobotLocomotion/drake/blob/master/drake/matlab/systems/plants/createKinematicsCacheAutoDiffmex.cpp#L26) (similar to the now `ToAutoDiffXd()`). Also notice how in that code [the `type_id` idea is used](https://github.com/RobotLocomotion/drake/blob/master/drake/matlab/systems/plants/createKinematicsCacheAutoDiffmex.cpp#L30) to instantiate the appropriate "Alternate".

These similarities on the designs bring to my mind the quote "great minds think alike".

RussTedrake, 2016-11-06:
@amcastro-tri -- the ToAutoDiffXd() (and ToSymbolicExpression) have been ordained as the way to go.  We've been writing code against them.  In fact, I was going to ask you about implementing ToAutoDiffXd() for RBT/RBP.  As soon as those exist, we'll be able to start using them for LQR and trajectory optimization.  :)

amcastro-tri, 2016-11-07:
Thank you for the update @RussTedrake. I created #4044 and updated #3987 to include it. I think this will be needed very soon in order to complete the checklist in #3987.

david-german-tri, 2016-11-22:
Linking this to #3967

sherm1, 2018-08-14:
This is obsolete now.

closed 2018-08-14 by sherm1