RobotLocomotion/drake #2849: can we system install swig instead of including it as an external?

RussTedrake, 2016-07-20:
We are currently building 3.0.9 from source.  For me (using make on mac), it is far and away the most painful part of the build process -- simply watching the printouts for the installation checks takes a non-negligible portion of my build time.

a few things have changed since we made the original decision:
 1) our list of officially supported platforms http://drake.mit.edu/developers.html#id6 has compressed and been upgraded
 2) homebrew, etc, is bottling newer versions (I see 3.0.10 bottled on my mac)
i'm wondering if we can revisit the decision and just apt-get install / brew install swig instead of building it from source?

@rdeits -- can you remind us on the initial pain point that forced you to include it as an external originally?

rdeits, 2016-07-20:
Yeah, the only reason we're building it from source is to get Matlab support, which is only available in a fork of the project.

RussTedrake, 2016-07-20:
so if we build swig_matlab, then we have to build swig?

jamiesnape, 2016-07-20:
Aside: For a while we used system SWIG when MATLAB was not being used, but apparently the superbuild is also a newer version of SWIG.

jwnimmer-tri, 2016-07-20:
Yes.  The swig syntax and capabilities between system swig and swig_matlab were divergent, so we switched to always using swig_matlab -- even for builds that didn't use matlab -- so that we only had to debug one set of `%.i` files.

RussTedrake, 2016-07-20:
that all makes sense.  just to confirm, is it not possible to build swig_matlab against the system installed swig, if the version is sufficiently up to date?

rdeits, 2016-07-20:
No, I don't believe so. swig_matlab is a fork of the swig source, not something that links against the swig libraries.

RussTedrake, 2016-07-20:
got it.  thanks.

closed 2016-07-20 by RussTedrake

rdeits, 2016-07-21:
It is worth considering, though, that as far as I'm aware there are exactly zero users of the Matlab swig bindings (but please correct me if I'm wrong). So if building `swig_matlab` is a major pain point, we can also consider just dropping those bindings entirely and going back to system swig for python (or really any of the other ways of generating python bindings: the deciding factor in my choice of swig over other options was the ability to also generate matlab bindings at the same time).

jwnimmer-tri, 2016-07-21:
Certainly if we can drop swig-matlab support, then I'd heartily recommend a non-swig interface generator for the python side.  Swig is less-than-ideal, and its only saving grace was multi-language support.

sherm1, 2016-07-21:
I understood the long term plan to be to hollow out the existing Matlab code and replace it by a thin wrapper around the new C++ code. If that's right then Swig's multi-language support could still be useful.

jwnimmer-tri, 2016-07-21:
That would depends on if we want to provide our matlab interface via mex, or swig, yes?

sherm1, 2016-07-21:
> That would depend on if we want to provide our matlab interface via mex, or swig, yes?

I'm ignorant of swig_matlab's wrapping approach. Does it write mex files? Matlab can also invoke Java bindings -- I've seen ordinary (non-matlab) swig used to interface to Matlab that way, with functional-but-ugly results.

rdeits, 2016-07-21:
@sherm1 yes, that has definitely been the plan, and was exactly why I picked `swig_matlab`. But I'm also willing to revisit the issue if it's making our lives much harder. 

`swig_matlab`produces a mex file for each swig module (`.i` file), and matlab `.m` files that call into that mex library for the wrapped functions and classes. It's similar to the way swig-python produces C extensions and thin `.py` wrappers. 

I don't believe I ever tried interfacing matlab to swig-java, but I agree that it would be possible. My experience has been that matlab <-> java performance was poor (specifically, converting java Strings to matlab char arrays was hilariously slow), so `swig_matlab`'s direct mex interface seemed like a better option.

RussTedrake, 2016-07-21:
Agreed.  The matlab-java interface is extremely convenient, but too slow to be the primary solution.