RobotLocomotion/drake #2852: Turn on -Werror=shadow checking

jwnimmer-tri, 2016-07-20:
Drake should turn on `-Werror=shadow` checking for compilers that support it.  Shadowing is extremely confusing and misleading, and we should never do it.

(Motivated by reviewing an PR yesterday that shadowed `int error` and lost the error code.)

jwnimmer-tri, 2016-07-20:
/CC @rpoyner-tri in case you are feeling generous

rpoyner-tri, 2016-07-20:
Early results with g++ 4.9 are disappointing: any case where a constructor argument has the same name as an accessor method, you get an error. These can be worked around, but it's worth considering whether "fixing" all such cases is too high a price to pay.

rpoyner-tri, 2016-07-20:
c.f.: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=57709

rpoyner-tri, 2016-07-20:
Still TODO (after #2867), remove excuses from these files:
- [ ] drake/examples/CMakeLists.txt
- [ ] drake/solvers/CMakeLists.txt
- [ ] drake/systems/controllers/CMakeLists.txt
- [ ] drake/systems/plants/CMakeLists.txt
- [ ] drake/systems/robotInterfaces/CMakeLists.txt
- [ ] drake/systems/test/CMakeLists.txt
- [ ] drake/systems/trajectories/CMakeLists.txt
- [ ] drake/util/CMakeLists.txt

The above list is probably not in the correct order of attack, but it is complete. In particular, the opt-out in drake/bindings/swig will never be removed; it protects auto-generated code.

rpoyner-tri, 2016-07-27:
With the merge of #2956, this issue is closed.

closed 2016-07-27 by rpoyner-tri