RobotLocomotion/drake #2865: file INSTALL cannot copy file liblcmtypes_bot2-core.a

jamiesnape, 2016-07-20:
https://drake-cdash.csail.mit.edu/viewBuildError.php?onlydeltap&buildid=55146

```
CMake Error at lcmtypes/cmake_install.cmake:117 (file):
  file INSTALL cannot copy file
  "/Users/elcapitan/workspace/mac-clang-continuous-release/pod-build/externals/bot_core_lcmtypes/lcmtypes/liblcmtypes_bot2-core.a"
  to
  "/Users/elcapitan/workspace/mac-clang-continuous-release/build/lib/liblcmtypes_bot2-core.a".
Call Stack (most recent call first):
  cmake_install.cmake:32 (include)

make[3]: *** [install] Error 1
```

mwoehlke-kitware, 2016-07-20:
I have *no idea* what would cause this... it looks like a strange system error (out of disk space?).

mwoehlke-kitware, 2016-07-25:
Is there a(n accessible) machine that is reproducing this reliably?

jamiesnape, 2016-07-25:
Not as far as I know.

mwoehlke-kitware, 2016-07-26:
Oh! Actually, I have a theory for what causes this and similar errors (e.g. "file too small (length=0) file '.../lib/liblcmtypes_bot2-core.a' for architecture x86_64"). Both libbot and bot2_core_lcmtypes generate this library. In parallel builds, it may be that they are tripping over each other in various ways. Probably we should either not build both, or make one depend on the other to at least force them to not be parallel. (Eventually I'd like to get rid of monolithic libbot entirely.)

jwnimmer-tri, 2016-07-27:
We are getting unlucky, and seeing this error more often locally now.  Adding a dependency from one to the other seems like an expedient and fair work-around.

PR #3058 (2016-08-04): Address libbot / bot_core_lcmtypes conflict
Both libbot and bot_core_lcmtypes generate bindings for the libbot LCM types. Since this means that there is significant overlap in the output files, including the C library, this can result in odd, apparently random errors if the two externals are built in parallel. Address this by adding an artificial dependency between the two to force them to be built in serial.

(In the long run, we hope to split lcmgl from libbot, and build only that and the stand-alone types package, dropping the monolithic libbot package entirely.)

Fixes #2865, hopefully.

@ [... cut]
commit f1bd5f37 (2016-08-04): Address libbot / bot_core_lcmtypes conflict

Both libbot and bot_core_lcmtypes generate bindings for the libbot LCM
types. Since this means that there is significant overlap in the output
files, including the C library, this can result in odd, apparently
random errors if the two externals are built in parallel. Address this
by adding an artificial dependency between the two to force them to be
built in serial.

(In the long run, we hope to split lcmgl from libbot, and build only
that and the stand-alone types package, dropping the monolithic libbot
package entirely.)

Fixes #2865, hopefully.
closed 2016-08-04 by mwoehlke-kitware