RobotLocomotion/drake #2869: DLL artifacts should go in `bin` on Windows, not `lib`

sherm1, 2016-07-20:
Windows expects `.dll`s (its version of shared libraries) to be collocated with executables (`.exe`s) and searches for them first, in the directory containing the referencing `.exe`, and second, via the same `PATH` environment variable that is used to locate executables. 

CMake has support for splitting generated `.lib` and `.dll` files into lib and bin directories, resp. It appears that we are using that only for LCM, so Drake's `install/bin` directory has just a single `lcm.dll` in it, while all the other several dozen Drake and external dlls are in `install/lib`.

Currently both `install/bin` and `install/lib` have to be added to `PATH` to get Drake's LCM examples to run on Windows, as well as `drake/lib` to find the developer's dlls. It is not good practice to have lib directories in the `PATH` variable that is intended to locate executables and dlls.

Moving just `lcm.dll` over to the lib directory likely won't work because LCM also has executable programs in that same directory that are likely expecting to find `lcm.dll` collocated. I think the best fix would just be to let CMake split the lib and dll files in the customary Windows fashion.

See discussions in PR #2854 and issue #2535.

david-german-tri, 2016-07-20:
The PR that fixes this should run and pass one of the Windows + MATLAB CI builds before merging.  `mex` files dynamically link some C++ dependencies, and there is special MATLAB magic in `addpath_drake` (and perhaps other MATLAB code) that configures the search path.

david-german-tri, 2016-10-06:
Windows is no longer supported.

closed 2016-10-06 by david-german-tri