RobotLocomotion/drake #2872: Move *SystemInterface out of the System2 API

david-german-tri, 2016-07-20:
Per discussion with @sherm1 and @RussTedrake, we will instead have a `System` class from which all Systems, including Diagrams, inherit.  There will be default implementations of methods that don't necessarily apply to all Systems - i.e. everything that is not part of the current `SystemInterface`.

For clarity, and just in case we change our minds later, I plan to keep `ContinuousSystemInterface` et. al. around as parents of `System`, as suggested by @jwnimmer-tri.  This will have no impact on implementers, and will make the transition more gradual as a side benefit.

david-german-tri, 2016-07-22:
As discussed on #2846, the `System` class will have (among other things) a concrete implementation of `get_name`.

amcastro-tri, 2016-07-27:
As I commented in #2955, why is this "low" instead of "high" priority given these two issues define the very basic API?

david-german-tri, 2016-07-27:
As a superficial refactor, it's low-priority relative to substantial new System2 features such as diagrams, state, cache, and non-numeric values, all of which are also API-changing.

sherm1, 2016-07-28:
Discussed w/@amcastro-tri today: working with the new API and providing real-user feedback is probably the most valuable part of his System 2 work. We would like to make these API-changing issues high priority if possible.

david-german-tri, 2016-07-29:
Adjusted to `priority:high` since it turns out this is blocking for Diagram state (or at least, the alternatives would be really ugly).

PR #2991 (2016-08-02): Introduce System<T>
This abstract non-interface base class template specifies all the methods that any System might implement. Functions that consume Systems should detect the System capabilities by inspection at runtime.

Resolves #2872.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2991)
closed 2016-08-02 by david-german-tri