RobotLocomotion/drake #2890: In System2, replace EvalOutputs with EvalOutput(index)

david-german-tri, 2016-07-22:
This will be useful to avoid computing unused outputs.  See discussion on #2846.

cc @sherm1

liangfok, 2016-07-22:
Shouldn't the implementation of `EvalOutputs()` be smart enough to know which outputs need to be evaluated? In other words, can the details of which outputs are actually evaluated be hidden from the caller?

david-german-tri, 2016-07-22:
@liangfok Right now, there's no way for the caller to say "I'm only going to use one output, so just compute that one."  This issue proposes to add that feature.  If a System has ReallyCheapOutput and ReallyExpensiveOutput, and your downstream code only looks at the cheap output, you shouldn't have to compute the expensive one.

Are you suggesting that the implementation of `EvalOutputs` could somehow deduce which outputs are important to the caller?  How would that work?

liangfok, 2016-07-22:
If there exists logic that determines whether downstream systems use a particular system's output, could that logic be included within the call to `EvalOutput ()`?

david-german-tri, 2016-07-22:
> If there exists logic that determines whether downstream systems use a particular system's output, could that logic be included within the call to EvalOutput ()?

If we're talking about Diagrams - sure, that's possible, if we broke encapsulation and let the System inspect the Diagram that contains it.  That would mean more code and tighter coupling to implement the same API with the same efficiency.

If we're talking about the general case where user code constructs a System, feeds it inputs, and inspects some of the outputs, I think the answer is self-evident?

jwnimmer-tri, 2016-07-22:
In the (common, likely) case that the computation for one output port shares common intermediate internal results with other output ports, how would this new design support sharing that intermediate computation?

david-german-tri, 2016-07-22:
Expensive intermediate data should be stored in a cache line.

liangfok, 2016-07-22:
Is the determination of whether a particular output consumed done once during initialization time or can it continuously change during run time? If the former, couldn't some static analysis be done to determine which outputs should be enabled during initialization?

Are we in danger of premature optimization?

david-german-tri, 2016-07-22:
> Is the determination of whether a particular output consumed done once during initialization time or can it continuously change during run time?

The latter.  For instance, at some sampling times only discrete ports are interesting to compute.

> Are we in danger of premature optimization?

This is not a costly or complicated change.  The only reason I opened this issue instead of just adding the change to #2846 is that #2846 is already large.

liangfok, 2016-07-22:
SGTM.

sherm1, 2016-07-22:
:+1: for this change. Note that this implies that we no longer tie side effects to `EvalOutput()` calls because a "sink" system like LCM publisher (#2821) has no output ports. We will need to add a method to the System interface that is reliably called at the right time regardless of the number of ports. Currently I've been thinking of it as a `Sample()` method but there is probably a better name. It would get called at the start of every (irreversible) time step for continuous Systems and at the sampling time for discrete Systems.

david-german-tri, 2016-07-22:
> Currently I've been thinking of it as a Sample() method but there is probably a better name. 

We're crossing over into #2836 territory here, but I still want to call it `Publish`.

sherm1, 2016-07-26:
:+1: for `Publish()`.

sherm1, 2016-07-28:
Can this be high priority (per discussion with @amcastro-tri)?

sherm1, 2016-12-22:
I'm working on this now as part of getting the cache working per #4364 and would like some feedback on the following approach:

OutputPorts need to be treated individually rather than as a group as we are doing now. Typically, each OutputPort will be backed by a cache entry that is evaluated when that port's value is requested, and brought up to date if necessary. The following per-port information must be available to manage the cache entry correctly:
- an allocator to provide an object that can hold the port's value
- a `Calc` method that computes the port's value into such an object
- the list of prerequisites to be used for invalidation.

I don't like the idea of collecting this information with methods that are indexed by port number. For systems with multiple output ports these methods would have to combine unrelated per-port computations and then branch internally on the port number (via brittle `switch` on constants, or some slower indirect method). That seems unnecessarily awkward. A better approach IMO would be to have an OutputPort object with virtual methods, so that the framework invokes `port.Eval(context)` rather than `EvalOutputPort(context, port_num)`. We are very close to having such an object already with SystemPortDescriptor; I propose to upgrade that to a first-class OutputPort object. Having an OutputPort object will help considerably in reducing our use of the ubiquitous and much-maligned port numbers.

I haven't worked out the details yet but want feedback on the general approach of using OutputPort objects to manage port values rather than using System-global methods indexed by port number. Does anyone think the port-numbered approach is preferable?

/cc @RussTedrake @david-german-tri @jwnimmer-tri @amcastro-tri

amcastro-tri, 2016-12-24:
> A better approach IMO would be to have an OutputPort object with virtual methods, so that the framework invokes port.Eval(context) rather than EvalOutputPort(context, port_num)

I really like this idea, I think it is pretty elegant. I do see how not only it'll help solve the problem of having some sort of `switch` in `EvalOutputPort()` but also it could help for a more intuitive way to declare ports by users. +1

RussTedrake, 2016-12-24:
I like the concept file, but do feel that code describing e.g. the derivative and output methods of the system should be conceptually close to each other in the code.  Of course, syntactic sugar like #4448 could be done in either case.  But it would be good to avoid having to define many classes to implement a simple input/output system.

amcastro-tri, 2016-12-24:
@sherm1, does your proposed design consider cases like `RigidBodyPlant`? `RigidBodyPlant` defines now input and output ports per model instance. At runtime, those outputs do different things depending on the port number. Would there be a single `OutputPort` class for all model instance input/output ports? how would the `OutputPort` decide what model instance it is supposed to evaluate?

sherm1, 2016-12-24:
@amcastro-tri, I would imagine a single class derived from OutputPort, say `InstanceOutputPort` that defines the information in a model instance output port. Then there would be `output_port` objects of concrete type `InstanceOutputPort` for each model instance. All the information needed to pick out the particular model instance (like its instance id) would have been supplied in the `InstanceOutputPort` constructor and kept with the object. Then when port's value is requested (by an input port that is plugged in to it), we invoke `port.Eval()` which extracts the appropriate instance's results, calling `port.Calc()` first if necessary to compute the results into the cache.

amcastro-tri, 2016-12-25:
I see, that makes sense. Thanks.

jwnimmer-tri, 2016-12-27:
To help frame the discussion, let's observe that `EvalOutputs` (now called `CalcOutput`) is called from _very few places_.  The main call comes from Digram; there's also one call coming from `Linearize` and one call coming from `LuenbergerObserver`.  So I think 3 in total now, so perhaps on the order of 10 call sites forever.

In contrast, there are dozens (if not hundreds) of _implementations_ of this method now (through the various `Do` NVI overrides, and the `Impl` flavor stashed away in automotive, etc).  So probably on the order of 1,000 (10,000?) call sites forever.

Thus, it matters _almost not at all_ at the call site what `port.Eval(context)` looks like versus `EvalOutputPort(context, port_num)`.

The only thing that really matters is what a user's specific `LeafSystem` subclass implementation looks like, and how easy it is to write such a class and have it be correct.

So following that line of thinking, for how many systems is it important to segregate port-specific computation into port-specific objects?  In the first place, only maybe 10% of our Systems are multi-ported to begin with.  Within those, how many of them could recapture a meaningful amount of compute resources by guarding computations with the desired port indices?  Clearly we can think of some RBP cases where it matters, but that is perhaps the exception, rather than the rule?

The posted proposal was to "upgrade SystemPortDescriptor to a first-class OutputPort object" but it wasn't clear if that new class is concrete (in which case user systems merely have to use it) or abstract (in which case user systems have to subclass it).  Given the priorities I listed above, if user code has to subclass anything other than `LeafSystem` in order to have a basic working System, we are probably doing it wrong.

In short: to assess a proposed port-specifier API, the main concern should be the effect on simple user LeafSystems (the 90% case), and the proposal didn't provide any illustrative code or prose to help us assess that impact.

RussTedrake, 2016-12-27:
I'm not sure that I agree.  You're extrapolating from very little data.  There are more implementations now because we're adding dynamical systems.  But I expect that the shift of our work will tend towards writing more algorithms (call sites forever >> 10).

jwnimmer-tri, 2016-12-27:
Fair enough.

So the proposal has to adequately address both caller and callee, instead of just the callee.

sherm1, 2016-12-27:
The caller side won't be much of a problem however this is done -- basically `port.Eval(context)` vs. `EvalOutputPort(context, port.get_port_number())`. The callee implementation side is very important but I don't think it is improved by using port numbers rather than ports, which is the question at hand. The interface for common cases (e.g. leaf system with a single vector output port) can be made nice either way. But I need to get the base System class and Context to support multiple cached and pass-through output ports of arbitrary type. To me that is a cry for a decent OutputPort abstraction, upon which much sweetness can be built. I think having some code to argue over is probably the best next step.

david-german-tri, 2017-01-03:
>  ...if user code has to subclass anything other than LeafSystem in order to have a basic working System, we are probably doing it wrong.

+1, for values of "basic" that include MIMO systems.  I also hope the code-to-argue-over preserves the const-Context, non-const-Output distinction.

PR #6038 (2017-06-14): Separate output ports for independent allocation and calculation
OutputPortDescriptor gets an upgrade to a full object called OutputPort, which supports three functions:
- allocate
- calculate (unconditional)
- evaluate (cached)

Resolves #2890 (along the lines proposed in [this comment](https://github.com/RobotLocomotion/drake/issues/2890#issuecomment-268889174)).
Resolves #4789.

Specifying an OutputPort requires specifying an allocator function and calculator function with matching types. The allocator can be provided as an explicit function or by providing a model value that can be copied  [... cut]
closed 2017-06-14 by sherm1