RobotLocomotion/drake #2906: Documentation for DircolTrajectoryOptimization?

aespielberg, 2016-07-24:
Hi all,

I'm trying to work on modifying DircolTrajectoryOptimization.  However, I am finding the constraint_fun as written a bit hard to parse.  I am wondering if there's any more detailed documentation of what the shared data pointers specifically point to?  I'm also wondering specifically what the variables xcol, dxcol, xdotcol, and dxdotcol are w.r.t the cubic interpolation?  Finally, is the Jacobian simply with respect to x0 and x1, or is it dependent on xdot0 and xdot1 in some way?

liangfok, 2016-07-24:
I assume that since this relates to constraints, @ggould-tri would be in the best position to answer this question. Please feel free to reassign as necessary.

hongkai-dai, 2016-07-24:
@liangfok @ggould-tri I wrote a lot of MATLAB code in DirectTrajectoryOptimization, I am happy to answer this one later tonight, if you do not mind.

liangfok, 2016-07-24:
Nice! That'll be much appreciated!

hongkai-dai, 2016-07-25:
@aespielberg , are you using the MATLAB interface? The `shared_data` in constraint_fun is introduced to cache the result and avoid redundant computation. For example, for a `RigidBodyManipulator`, computing its kinematics tree can be time consuming, so we cached the result in the `kinsol` object, as the return argument of `doKinematics` function. When we impose multiple kinematic constraints in trajectory optimization, these constraints can share the cached `kinsol` object, rather than computing the kinematics tree by their individual `constraint_fun` function. Also in `DirectTrajectoryOptimization`, we cache `xdot` and `dxdot`, since they are used for twice, when `xdot,dxdot` at i'th knot point is used in interpolating the [i,i+1] knot points, and [i-1,i] knot points.

The `xcol`, `dxcol`, etc are the value of the states at the collocation point. For more details on the math, you can refer to Hagraves paper "Direct trajectory optimization using nonlinear programming and collocation".

The Jacobian of the constraint is w.r.t `x0,x1,u0,u1`. We do not  explicitly include the Jacobian w.r.t `xdot0,xdot1` since they are not our decision variables. Implicitly, these Jacobian can be obtained by using `xdot=f(x,u)` and the Jacobian w.r.t `x,u`.

If you have further questions about `DirectTrajectoryOptimization`, please let us know. You can also talk with @mposa in our lab if we cannot reply you immediately.

ggould-tri, 2016-07-25:
Unfortunately I have no knowledge of the matlab optimizer stack, so I'm not going to be much help here.  But it sounds like Hongkai has the question well in hand :-)

hongkai-dai, 2016-07-28:
@aespielberg , did the comments answer your question? Shall we close this issue, or you would like further discussion on this thread?

aespielberg, 2016-07-28:
Thanks for the replies; I have been busy the last couple of days so I'm
going to look into it tomorrow, I will let you know if I have further
questions then.

Thanks!

hongkai-dai, 2017-03-13:
@aespielberg is it OK if we close this issue?

aespielberg, 2017-03-13:
Yes, sorry for forgetting to close it myself.  I do recommend adding
explicit documentation of the mathematics and representation (or references
to them) to future Drake releases though.

-Andy S.

hongkai-dai, 2017-03-13:
I totally agree on the documentation. We are rewriting the code in C++, and the documentation is much cleaner there.

closed 2017-03-13 by hongkai-dai