RobotLocomotion/drake #2915: OS X MATLAB tests failing with "Sorry, you have not run 'make' yet..."

jamiesnape, 2016-07-25:
```
Sorry, you have not run 'make' yet in the drake root,
which means you do not have the compiled MEX files needed to run this program.
Running configure and make in the drake root directory will fix this.
```

david-german-tri, 2016-07-25:
Link?

jamiesnape, 2016-07-25:
Added.

david-german-tri, 2016-07-25:
Specifically, this looks like the first red nightly.  Odd that it includes no CMake changes.

https://drake-jenkins.csail.mit.edu/view/Nightly/job/mac-clang-nightly-matlab/112/

The changes to drake-ci on July 22 also don't smell like culprits.  Interesting.

https://github.com/RobotLocomotion/drake-ci/commits/master

mwoehlke-kitware, 2016-07-25:
It appears that the real issue is in [`solvers/NonlinearProgramSnoptmex.cpp`](https://github.com/RobotLocomotion/drake/commits/master/drake/solvers/NonlinearProgramSnoptmex.cpp).

This also tripped CI in #2896. Not sure what changed that this is suddenly tripping, but it's a real build error and should probably be fixed by someone more familiar with the code in question than myself...

david-german-tri, 2016-07-25:
@rpoyner-tri FYI

mwoehlke-kitware, 2016-07-25:
Probably broke here: f2c4d53d0ed636cf482fe23e4ca0775e1a7e2333.

david-german-tri, 2016-07-25:
Good catch, @mwoehlke-kitware.  I guess it's a clang-specific error, since it didn't show up in Linux+gcc+MATLAB.

mwoehlke-kitware, 2016-07-25:
I believe `-Wshadow` is not enabled under GCC; see [`drake/CMakeLists.txt:83`](https://github.com/robotlocomotion/drake/blob/830d4766359c557e24cb5519e82ba8e39af353a3/drake/CMakeLists.txt#L83)

david-german-tri, 2016-07-25:
@rpoyner-tri, over to you.  Looks like Mac MATLAB tests should be requested from Jenkins for `-Wshadow` cleanup.

rpoyner-tri, 2016-07-25:
@jamiesnape: help me out here. I've submitted #2920 and requested a mac clang matlab build, but I see Jenkins telling me that all 'mac' nodes are offline. Is that something that should worry me?

jamiesnape, 2016-07-25:
That is normal. They need time to boot and establish a connection to Jenkins master (2-3 minutes).

rpoyner-tri, 2016-07-25:
closed via #2920.

closed 2016-07-25 by rpoyner-tri