RobotLocomotion/drake #2924: Every continuous build is failing during LCM checkout

jamiesnape, 2016-07-25:
```
fatal: reference is not a tree: 9c9e22cf23663ef9cd0e5ec83932ff7f70cd7b18
Unable to checkout '9c9e22cf23663ef9cd0e5ec83932ff7f70cd7b18' in submodule path 'externals/lcm'
```

https://drake-jenkins.csail.mit.edu/view/Continuous/

liangfok, 2016-07-25:
Experimental builds also appear to be having this problem:

https://drake-jenkins.csail.mit.edu/job/experimental/3026/

rpoyner-tri, 2016-07-25:
Looks to me like the hash we think we want has disappeared from upstream: 
```
$ git clone git@github.com:mwoehlke-kitware/lcm.git kitware-lcm
Cloning into 'kitware-lcm'...
remote: Counting objects: 6783, done.
remote: Compressing objects: 100% (6/6), done.
remote: Total 6783 (delta 0), reused 0 (delta 0), pack-reused 6777
Receiving objects: 100% (6783/6783), 7.79 MiB | 4.51 MiB/s, done.
Resolving deltas: 100% (4278/4278), done.
Checking connectivity... done.
$ cd kitware-lcm/
$ git show 9c9e22cf23663ef9cd0e5ec83932ff7f70cd7b18
fatal: bad object 9c9e22cf23663ef9cd0e5ec83932ff7f70cd7b18
```

closed 2016-07-25 by jamiesnape

mwoehlke-kitware, 2016-07-26:
Sorry! The SHA should be back now; see #2928 for long explanation...

PR #2928 (2016-07-25): Update lcm submodule
Fixes #2924.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2928)