RobotLocomotion/drake #2926: error when building drake as a ROS catkin package

hongkai-dai, 2016-07-25:
I followed the instruction on http://drake.mit.edu/from_source_ros.html#build-from-source-using-ros-catkin to build drake using catkin. But I get the following message after I call
```
catkin build -DCMAKE_BUILD_TYPE:STRING=RelWithDebInfo
```

```
Errors     << drake_cars_examples:make /home/hongkai/dev/drake_catkin_workspace/logs/drake_cars_examples/build.make.001.log
In file included from /home/hongkai/dev/drake_catkin_workspace/src/drake_ros_integration/drake_cars_examples/src/car_sim_lcm_and_ros.cc:10:0:
/home/hongkai/dev/drake_catkin_workspace/src/drake_ros_integration/drake_ros_systems/include/drake_ros/ros_vehicle_system.h:9:50: fatal error: ackermann_msgs/AckermannDriveStamped.h: No such file or directory
#include "ackermann_msgs/AckermannDriveStamped.h"
                                                 ^
compilation terminated.
make[2]: *** [CMakeFiles/car_sim_lcm_and_ros.dir/src/car_sim_lcm_and_ros.cc.o] Error 1
make[1]: *** [CMakeFiles/car_sim_lcm_and_ros.dir/all] Error 2
make: *** [all] Error 2
cd /home/hongkai/dev/drake_catkin_workspace/build/drake_cars_examples; catkin build --get-env drake_cars_examples | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
.........................................................................................................................................
Failed     << drake_cars_examples:make           [ Exited with code 2 ]
Failed    <<< drake_cars_examples                [ 4.7 seconds ]
[build] Summary: 2 of 3 packages succeeded.
[build]   Ignored:   1 packages were skipped or are blacklisted.
[build]   Warnings:  1 packages succeeded with warnings.
[build]   Abandoned: None.
[build]   Failed:    1 packages failed.
[build] Runtime: 1 minute and 27.9 seconds total.
```

liangfok, 2016-07-25:
Try installing the following package:

```
sudo apt-get install ros-indigo-ackermann-msgs
```

I see that it is listed towards the bottom of the linked page. I'll move it up in a forthcoming PR.

hongkai-dai, 2016-07-25:
Thanks for the prompt reply! The installation is successful now!

PR #2927 (2016-07-25): Update install instructions for ros-indigo-ackermann-msgs.
Moved instructions to install `ros-indigo-ackermann-msgs` to the top of the Drake/ROS install section.

Resolves #2926.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2927)
closed 2016-07-25 by sherm1