RobotLocomotion/drake #2930: remove tspan from RigidBodyConstraint

hongkai-dai, 2016-07-25:
The `tspan` field in the `RigidBodyConstraint` was introduced to work with `inverseKinTraj`, such that the `inverseKinTraj` can determine whether a constraint is active or not. I think this `tspan` was a bad design choice, for several reasons

1. It is only useful in the `inverseKinTraj` function. The `RigidBodyConstraint` itself should not depend on time. Even if we need any timing information, this information should be handled by `inverseKinTraj`, not `RigidBodyConstraint`.
2. It is problematic to determine whether a constraint is active or not by using a time span, as we need to compare if a given time falls within the time span. Comparing the value of doubles is always a bad choice.
3. `inverseKinTraj` should directly determine if a constraint is active or not for each knot point. When adding that constraint to `inverseKinTraj`, we should add an input argument `active_knots`, I feel this is more clear than using a time span and check if the knot falls within the time span.

I think currently we should not worry about the "inbetween time samples" when using `inverseKinTraj`. As far as I knot, nobody is using this feature.

@sammy-tri , what do you think? If it is OK withi you, I can start working on this.

sammy-tri, 2016-07-26:
I think it's probably a good idea, but I've got a pull request (well, two I think) coming this week to refactor inverseKinTraj to use the new OptimizationProblem API, so it's probably not worth doing on the existing direct-to-SNOPT implementation.

I think inverseKinPointwise also uses isTimeValid and tspan to determine when a constraint is active?

I implemented the inbetween time samples on my refactor...   I could definitely have saved some time had I not bothered.  :)

hongkai-dai, 2018-11-01:
As we move to use `InverseKinematics`, and the new type of constraint in https://github.com/RobotLocomotion/drake/blob/master/multibody/inverse_kinematics/position_constraint.h, we will abandon `RigidBodyConstraint` soon.

hongkai-dai, 2019-03-04:
Close this as we will deprecate RigidBodyConstraint.

closed 2019-03-04 by hongkai-dai