RobotLocomotion/drake #2931: support removing constraints from optimizationproblem

hongkai-dai, 2016-07-25:
Currently we cannot remove a constraint from optimization problem. This feature will be very useful when we need to solve a sequence of similar optimization problems, like in model predictive control. 

One approach is to keep an ID of the constraint when being add, and remove the constraint using that ID.  Conceptually, we can add an `constraint_ID_list` field in `OptimizationProblem`

```
OptimazationProblem* prog = new OptimizationProblem() // prog->constraint_ID_list is empty
int constraint_id = prog->addGenericConstraint(constraint_object); // prog->constraint_ID_list contains constraint_id
prog->removeConstraintByID(constraint_id); // Look up the constraint_ID_list with the matching ID, remove the constraint in its list with the same index, and determine the type of the problem. Now prog->constraint_ID_list is empty
```

What do people think of this?

sherm1, 2016-07-25:
Seems reasonable to me. It might have some implications for how constraint objects are stored in the `OptimizationProblem` object. For example, it would preclude storing them in an `std::vector` indexed by constraint id since this would create holes (or at least holes would have to be allowed). Iterating through the constraint objects would require something fancier than `for (int id=0; id < get_num_constraints(); ++id) {}`

BTW (unrelated to your question) there is no need to heap allocate objects in C++. The line
```c++
OptimizationProblem* prog = new OptimizationProblem();
```
is Java-flavored. The C++ version should normally be:
```c++
OptimizationProblem prog;
```

hongkai-dai, 2016-07-25:
@sherm1 , thanks! I was thinking not to index the constraint by id. And there will be two lists stored in `OptimizationProblem`, the `constraint_ID_list` and the `constraint_list`. When we want to remove a constraint with given ID `remove_ID`, we can do
```
auto constraint_it = constraint_list.begin();
auto id_it = constraint_ID_list.begin();
for(;id_it != constraint_ID_list.end();id_it++) {
  if(*id_it == remove_ID) {
  constraint_ID_list->erase(id_it);
  constraint_list->erase(constraint_id);
  }
  else {
    constraint_it++;
  }
}
```
This should also work for vectors. This should avoid the hole in the containers.

Thanks for pointing out the heap allocation problem!

jwnimmer-tri, 2016-07-26:
I disagree with @sherm1.  Using bare integer IDs does not support type-checking or encapsulation very well, and we should do our best to avoid them in new code.

In my opinion, the cleanest API will be value based -- you do `add_constraint(foo)` and then `remove_contraint(foo)`.  Is that possible here?  It doesn't even have be the same object -- it could be a new instance that compares as equal, perhaps?

Alternatively, items could have string names and the calling code could have `kFooName` that it uses to refer to the elements of the problem (to add and remove).

Alternatively, return opaque handles instead of bare ints.

Alternatively, is the intent to remove the constraint, or just disable it?  If just disabling, then perhaps a method on the constraint itself would suffice.

sammy-tri, 2016-07-26:
I'm with @jwnimmer-tri that we should not use bare integer ID's to handle removing constraints.  Fortunately, I think doing so would only make out lives harder anyway...

Currently when a Constraint is added to an OptimizationProblem, there are two basic forms of the interface -- one where a shared_ptr to the Constraint is passed to OptimizationProblem (returns void) and one where the constraint parameters are passed to OptimzationProblem and a shared_ptr to the Constraint is returned.  Either way, OptimizationProblem internally stores an instance of OptimizationProblem::Binding which holds the shared_ptr.  I think the shared_ptr makes sense as the handle to pass to the function which removes the Constraint from the OptimizationProblem. (it's already stored and already available to the client).

OptimizationProblem currently stores the known constraints using std::list.  I don't see a reason to change that.

jwnimmer-tri, 2022-05-02:
Fixed in #15350.

closed 2022-05-02 by jwnimmer-tri