RobotLocomotion/drake #2949: Sphinx Warnings When Rendering CLion.rst

liangfok, 2016-07-27:
# The Warnings

```
/home/liang/dev/drake-distro-2/drake/doc/clion.rst:24: WARNING: Enumerated list ends without a blank line; unexpected unindent.
/home/liang/dev/drake-distro-2/drake/doc/clion.rst:28: WARNING: Bullet list ends without a blank line; unexpected unindent.
/home/liang/dev/drake-distro-2/drake/doc/clion.rst:29: WARNING: Block quote ends without a blank line; unexpected unindent.
```

amcastro-tri, 2016-07-28:
How can I reproduce this warning @liangfok?

liangfok, 2016-07-28:
Here are the commands assuming you're using `make` (replace `make` with `ninja` if you're using `ninja):

```
cd drake-distro/build/drake/
rm -rf doc
make documentation
```

amcastro-tri, 2016-07-29:
@liangfok. With `ninja` I get:

`
FAILED: cd /home/amcastro/Drake/Master/drake-distro/build/drake/doc && /usr/local/bin/sphinx-build -b html -c /home/amcastro/Drake/Master/drake-distro/build/drake/doc -d /home/amcastro/Drake/Master/drake-distro/build/drake/doc/doctrees /home/amcastro/Drake/Master/drake-distro/drake/doc /home/amcastro/Drake/Master/drake-distro/build/drake/doc/sphinx
Error: Config directory doesn't contain a conf.py file.
ninja: build stopped: subcommand failed.
`

liangfok, 2016-07-29:
Based on @jwnimmer-tri's suggestion, I updated the reproduction instructions to use `touch` instead of `rm`. @jwnimmer-tri was able to reproduce the warning using `ninja`.

amcastro-tri, 2016-07-29:
I am able to reproduce the same warnings. If I remove (or add) CR's as requested by the warnings (that's what I understand), the warnings go away. However, the parsed documentation does not look right. 

The problem here is that I have a bullet list nested inside a numbered list. Maybe sphinx doesn't like that? ideas?

liangfok, 2016-07-29:
I recommend figuring out how to supress the warnings since the existing `clion.rst` renders OK.

jwnimmer-tri, 2016-07-29:
Eep, no!  Just figure out the correct `rst` syntax from the ReST specification / tutorial / documentation.

liangfok, 2016-08-04:
FTR, this warning is still present: https://drake-cdash.csail.mit.edu/viewBuildError.php?type=1&buildid=69086

PR #3153 (2016-08-16): Clean up rst formatting for clion
- Fix indentation of bullet list to line up with the above (fixes #2949).
- Replace markwdown literals (backtick) with rst literals (double-backtick).
- Line wrap to fit in 80 columns.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/3153)
commit a5b74da3 (2016-08-16): Clean up rst formatting for clion
- Fix indentation of bullet list to line up with the above (fixes #2949).
- Replace markwdown literals (backtick) with rst literals (double-backtick).
- Line wrap to fit in 80 columns.
closed 2016-08-16 by jwnimmer-tri