RobotLocomotion/drake #2953: examples/Quadrotor/createFunnelLibrary failes to run

baxelrod, 2016-07-27:
There seem to be a number of issues related to the createFunnelLibrary example in the Quadrotor folder. I'm using a source build, so I think we do need to check that it is not my configuration that is at fault. 

It fails at several points. 
1) numel(obj.robot.getStateFrame.coordinates) doesn't work for the type of frame used in the example (this appears in two files). 

Sidestepping the issue (using the dim property), yields the following error later in execution.

2) sysClPoly = taylorApprox(sysCl, xtraj, [], 3);  FunnelLibrary/computeFunnel
sysCl seems to be a Simulink object and cannot be taylor approximated. 

Are you able to replicate these issues simply by running the createFunnelLibrary? 

Thanks,
Brian

baxelrod, 2016-07-29:
A bit more information. All this occurs within drake/systems/trajectories/FunnelLibrary.m

Line 48 returns a simulink object which doesn't work with line 50. 

Looking further, I get the following error message
```feedback combinations with saturations were causing problems in simulation.  See https://github.com/RobotLocomotion/drake/issues/494```

Would turning off the input saturation constraint in the model be the only easy way to fix this?

baxelrod, 2016-07-30:
Turning off input saturation leads to more errors. 

Am I running an unfinished example?

hongkai-dai, 2016-07-30:
I can resolve the first issue by replacing `numel(obj.robot.getStateFrame.coordinates)` with `obj.robot.getNumStates`. @psiorx , do you know what caused the second problem?

baxelrod, 2016-08-01:
@hongkai-dai , @psiorx The second problem is due to how input saturation is handled. It's handled through simulink, which doesn't allow for taylor approximation. 

Turning off the input saturation fixes the issue, but reveals further issues down the pipeline (which I'm happy to help you debug). 

@hongkai-dai Do you mind having a brief chat offline about what parts of Drake I should expect to work?

hongkai-dai, 2016-08-01:
@baxelrod , sounds great. Where are you? I should be back to Cambridge tomorrow, and stay there for the rest of the week. Or you can send me email at daihongkai@gmail.com

psiorx, 2016-08-01:
@hongkai-dai @baxelrod 
Taking a look at this.  It looks like some of the changes to FeedbackSystem and MultiCoordinateFrame may have broken this. It takes about 5-7 hours to run so it hasn't been getting test coverage but I'll try to find the point where problems were introduced.

@baxelrod If you need this right away, I've confirmed that the original commit (cc9352) still works.

baxelrod, 2016-08-01:
@psiorx Thanks! I'll use that commit for now. 

@hongkai-dai I'm actually on the west coast (Microsoft Research Redmond/Seattle). I think I'll send you an email with questions in a bit.

psiorx, 2016-08-01:
@hongkai-dai 
>  do you know what caused the second problem?

In the code that works, the feedback combination of a `RigidBodyManipulator` and an `AffineSystem`  results in a `FeedbackSystem` which is amenable to Taylor approximation.  However, in the latest code this feedback combination results in a `SimulinkModel` for which the `DrakeSystem.taylorApprox` no longer works.  

Trying to pinpoint when that got changed.  I'll update when I find it.

baxelrod, 2016-08-01:
@psiorx The feedback function produces a simulinkmodel when it detects input limits. If you read one of the warnings, it cites the commit that breaks it.

psiorx, 2016-08-01:
@baxelrod 

> Turning off the input saturation fixes the issue, but reveals further issues down the pipeline (which I'm happy to help you debug). 

Commenting out the check on lines 25-27 in FeedbackSystem.m fixes this for me.  Is this what you did when you hit further problems downstream?

baxelrod, 2016-08-01:
Nope, but I did something with a similar effect. 

Do you not have problems downstream?

psiorx, 2016-08-01:
> Do you not have problems downstream?

It has been running for over an hour now and seems to be doing fine.  I'll let it run overnight and post an update in the morning.

It does require both Mosek and Sedumi solvers.  I was not able to get around the numerical instabilities with just a single solver.

baxelrod, 2016-08-01:
Also a little sidenote. 

I think the matlabpool function has been deprecated in the latest version of matlab. 

Error in PolynomialSystem/sampledFiniteTimeReachabilityFunnel>findL (line 310).

psiorx, 2016-08-02:
The script ran into a Mosek licensing issue midway through.  Waiting on an updated license email.  I do have the finished finished funnel `.mat` file that I could throw on dropbox if you'd like to try out the online funnel planning example.

> the matlabpool function has been deprecated in the latest version of matlab

Thanks for the heads up.  I'll remove it.

baxelrod, 2016-08-02:
It still doesn't run for me, using the commit you mentioned, and mosek and sedumi. 

It crashes not being able to generate one of the funnels. I presume that since it is an example, it should pass? (no uncertifiable trajectories are being generated).

baxelrod, 2016-08-02:
@psiorx Does it go until completion for you? 

I'm hoping to use the functions in the example, I'm not particularly interested in the exact output of the example. I'm hesitant the functions unless I can get an example to work.

psiorx, 2016-08-03:
@baxelrod 

I just ran this to completion on a slightly modified(disabled input limits before the funnel computation) Drake master branch after fixing the Mosek licensing issues.

This is the figure that should pop up if everything is working correctly:

[image]

I'm using MATLAB R2014a with MOSEK Version 7.1.0.45 (Build date: 2016-1-25 06:52:46)

psiorx, 2016-08-03:
>  one of the funnels is generating an infeasible SOS/SDP

I've found the numerics on this algorithm to be very finicky.  Mosek will complain that a problem is infeasible and then sedumi is happy to solve it or vice versa.  Also, things that shouldn't change feasibility like uniformly scaling the objective function can sometimes make or break the solver's ability to handle it.

jwnimmer-tri, 2017-06-19:
Closing as won't-fix, because #6369 will be removing the relevant software.

If I'm confused and this should in fact stay open, please re-open an clarify the remaining unsolved problem(s) in a new comment here.

closed 2017-06-19 by jwnimmer-tri