RobotLocomotion/drake #2955: Separate PortDescriptor from InputPort/OutputPort in System2

david-german-tri, 2016-07-27:
As System2 is currently implemented, the `InputPort` class and `OutputPort` class serve two purposes.

1. They describe the metadata of the port: sample rate, vector vs abstract value, etc.

2. They contain the data of the port, whether directly or by reference.

@sherm1 and @amcastro-tri have proposed separating these two functions, so that a System can describe its I/O topology without ever allocating a Context or Output.  I plan to accomplish this by factoring the metadata out into a `PortDescriptor`.  Additionally, the `PortDescriptor` will have a pointer to its system, and a unique identifier within the system, so that it's possible to connect two `PortDescriptor`s in a Diagram with no other arguments.

This should not be worked until after #2872 is resolved.

amcastro-tri, 2016-07-27:
I would actually make `InputPort` and `OutputPort` to be these descriptors you are talking about so that they contain the connectivity of the system. Then the output data in the context can simply be a vector (or you can encapsulate it in some `OutputPortData` class if you will. 

I like exposing `InputPort` and `OutputPort` as the basic public API's to the user so that we can define input/outputs as commented in the discussion in #2846 here [in this comment](https://reviewable.io/reviews/robotlocomotion/drake/2846#-KNYrHOybERiOuVxgaYN).

amcastro-tri, 2016-07-27:
Also, why are #2872 and this issue "low" priority? shouldn't they be "high" in your list given they define the very basic API to our humble users? (like me :-) )

david-german-tri, 2016-07-27:
@amcastro-tri You will be able to say `diagram.Connect(const PortDescriptor& src, const PortDescriptor& dest)` under the design I proposed above.  However, ports do not contain the connectivity.  Diagrams contain the connectivity.  This is important because a single instance of a System might be asked to evaluate Contexts with different InputPorts over its lifetime.

You are correct that this will change user-facing API.  This means it should be resolved before we actually include System2 in a Drake release.  However, it's a superficial change, and thus low-priority relative to building out core features.  Bleeding-edge users will have to deal with some churn.

amcastro-tri, 2016-07-27:
But I'd like better to say `spring_mass_system.output_port("Position").connect(pid_system.input_port("Input signal")`.
Where `System::output_port` returns an `OutputPort` and `System::input_port` would return an `InputPort`. These Input/Output ports only hold pointers to the port they connect to (and maybe some other meta-data like sizes, units and so forth). 
Then "global" pointers (more like `int` indexes) to the specific entries in the context pointed to by ports are setup when you call `Diagram::Finalize()`.

david-german-tri, 2016-07-27:
A principle of the System2 design is that systems do not know or care who produces their inputs, or who consumes their outputs.  This has some nice properties; for instance, it's easy to maintain multiple Contexts for the same system.

Of course, one could also write a perfectly good producer-consumer graph that doesn't follow this principle.  But let's be clear: you're asking me to scrap a significant piece of design.  Thus far, the only reason given is that you would like to write:

`a.output_port("foo").connect(b.input_port("bar"))`

whereas I am proposing to provide:

`diagram.connect(a.output_descriptor("foo"), b.input_descriptor("bar"))`

This does not seem like sufficient reason to me.

sherm1, 2016-07-27:
I would be happy with either of those APIs. I don't think @amcastro-tri was proposing a design change, assuming he was thinking of it as I did in System3, like this (in pseudocode):
```c++
diagram.AddSystem(a); // takes ownership of a
diagram.AddSystem(b); // takes ownership of b
a.output_port("foo").connect(b.input_port("bar"));
    // verifies that both a and b are in the same diagram, then invokes diagram.Connect()
// I think this is about as good, though:
diagram.Connect(a.output_port("foo"), b.input_port("bar"));
    // still has to verify that a and b belong to the diagram
```
This makes sense if you think of the diagram as owning its subsystems.

david-german-tri, 2016-07-27:
Ah, I see.  Yes, that would work for a diagram that owns its constituent systems, which I agree we should support.  However, I am still not convinced we should require that diagrams own their constituent systems, so I'd like to stick with the `diagram.Connect(a, b)` interface because it's neutral to the issue.

sherm1, 2016-07-28:
API change -- can this be high priority (per discussion with @amcastro-tri)?

david-german-tri, 2016-07-29:
I'd prefer to leave this at `priority:low`.  I agree it can be done, and I have no objection to doing it eventually, but it's just syntactic sugar and there's too much other work to be done.

As discussed on #2970, this feature will also be useful for System authors who _really really don't want_ to have customized input and output vector classes.  Those authors can instead declare ports in the constructor, and we can auto-allocate BasicVectors for them in the Context and Output.  I don't really understand why this is desirable, but @amcastro-tri has convinced me we can accommodate it without breaking anything.

sherm1, 2016-07-29:
Copying my comment from #2970 since it is probably more useful here:

> I agree it *may* be useful for Input/Output vector ports to have semantics while remaining differentiable (otherwise AbstractValue ports could have all the semantics anyone could possibly want). Anyway the VectorInterface class is sufficient for that and doesn't really complicate the API for ports since it is just a value type among many other possible value types.

> I don't agree that the same is valuable for derivative vectors. Those ought to match the semantics of their corresponding continuous states (in whole xc or in parts qdot,vdot,zdot, and qdotdot). I think the relationship between continuous states and their derivatives is plenty of semantics and that attempting to lay on more is confusing and unnecessary.

david-german-tri, 2016-08-02:
> derivative vectors ... ought to match the semantics of their corresponding continuous states 

I agree.  I would expect the derivative vector always to have the same leaf type as its corresponding continuous state, so `xc.get_foo()` is the value of `foo` and `xcdot.get_foo()` is the value of the derivative of `foo`.

PR #3102 (2016-08-15): Provide a mechanism for Systems to declare their input/output topology
This shouldn't be merged yet because it lacks test coverage, but I'd appreciate design feedback from @sherm1 and @amcastro-tri before I sink time into writing some.

Resolves #2955 (I hope).

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/3102)
closed 2016-08-15 by david-german-tri