RobotLocomotion/drake #2962: Consider adding "minimal externals" to supported configurations

jwnimmer-tri, 2016-07-28:
http://drake.mit.edu/developers.html lists our supported configurations.  For the "Superbuild Deps" we have both `"Default"` and `"Default + MATLAB"`.

Perhaps we should consider adding a `"Minimal"` supported configuration as well, where only `Eigen` and `googletest` are used.  Untested code is broken code -- and if we actually want to support use of Drake with a minimal footprint, we need to bring that under test.

When I tried turning off all the dependencies today as an experiment, I found several regressions: bullet (#2958), lcm (#2960), swig_matlab (#2961), and ipopt (RobotLocomotion/drake-ci#18).

What do folks think?

jwnimmer-tri, 2016-07-28:
/CC to @RussTedrake and @david-german-tri and @jamiesnape specifically for thoughts.

jamiesnape, 2016-07-28:
Sure. Debug or Release by default?

jwnimmer-tri, 2016-07-28:
For pre-merge testing I would propose Debug, to get the best diagnostic output.  Since many, many tests will be disabled, I suspect the runtime duration of the build would be alright.  For post-merge testing, we'd probably want to have both Debug and Release (at least on one easy target, such as ubuntu).

jamiesnape, 2016-07-29:
`WITH_SPDLOG` on or off for a minimal build?

sherm1, 2016-07-29:
I vote for spdlog "on".

jwnimmer-tri, 2016-07-29:
It should be off -- it is not required, and has specific code to stub spdlog when unavailable that we should be testing.

RussTedrake, 2016-09-08:
i definitely support.  but my view is that externals should (almost) always be built in release mode.  only drake should have to go into debug mode.

jwnimmer-tri, 2016-09-08:
I think #3092 fixed every question here.

closed 2016-09-08 by jwnimmer-tri