RobotLocomotion/drake #2966: Add Instructions on how to Run Drake + ROS CI builds

liangfok, 2016-07-28:
Update [from_source_ros.rst](https://github.com/RobotLocomotion/drake/blob/master/drake/doc/from_source_ros.rst) with details on how to trigger the Drake+ROS CI builds as exemplified [here](https://github.com/RobotLocomotion/drake/pull/2935#issuecomment-235252445).

jamiesnape, 2016-07-28:
All the "**drake-jenkins-bot** ...-experimental-... please" builds need documentation.

liangfok, 2016-09-29:
The instructions on how to schedule a CI test for Drake + ROS is available here: http://drake.mit.edu/from_source_ros.html#scheduling-a-drake-ros-continuous-integration-test

I created #3625 to track the documentation of _all_ the builds and will close this issue.

closed 2016-09-29 by liangfok