RobotLocomotion/drake #2968: Check Model Validity Each Simulation Cycle; Abort if Invalid

liangfok, 2016-07-28:
We should check that the positions and velocities of all joints in the simulation are valid each cycle of the simulation loop. This should result in the simulation terminating with a good error message (i.e., "joints foo, bar, etc., exceeds velocity limits").

Currently, the simulation "blows up" with a useless error message like:

```
terminate called after throwing an instance of 'std::runtime_error'
  what():  In BulletModel::findClosestPointsBetweenElements: No closest point found between 6935776 and 6939184

```

amcastro-tri, 2016-07-28:
@liangfok: could you provide the specific case that is blowing? is that one of our examples in master?
How could I reproduce this?

liangfok, 2016-07-29:
Basically, any time the simulation goes unstable, the simulation should terminate by printing joint velocity and position limit violation errors rather than a bullet collision modeling error. Here's one way to replicate the error in the car demo:

# Step 1 - Increase Car Demo Integrator Time Step Size by 10X

Modify `initial_step_size` to be `5e-2` here: https://github.com/RobotLocomotion/drake/blob/19bcddb4eaadc5dd4e1c695515abf2b9604f13c3/drake/examples/Cars/car_simulation.cc#L264

Re-build Drake.

# Step 2 - Run the Demo

Execute:

```
$ cd drake-distro/build
$ ./drake/bin/car_sim_lcm ../drake/examples/Cars/models/prius/prius_with_lidar.sdf ../drake/examples/Cars/models/stata_garage_p1.sdf 
```

sherm1, 2016-12-22:
@edrumwri, assigning to you to decide whether to keep this around or maybe replace it with something more specific. I interpret this as a general request for the Simulator to attempt to provide useful information about systems that blow up, which does seem like a good idea where possible. Some general things could be done like catching NaNs, determining which state drove the stepsize to epsilon, and catch/recover from derivative evaluation failures. Violation of constraints can be noticed at the Simulator level, but violation of internal restrictions such as the one described here probably have to be detected by the System being simulated.

david-german-tri, 2017-01-03:
> Violation of constraints can be noticed at the Simulator level, but violation of internal restrictions such as the one described here probably have to be detected by the System being simulated.

Which makes me wonder whether this issue is also a feature request for RigidBodyPlant in particular.

edrumwri, 2017-01-04:
I'm closing this issue because the robustness of the integration mechanisms are under active improvement, and I suspect that some of these instability issues will naturally disappear as error controlled integration becomes the default. I'm happy to re-open it if this problem becomes more prevalent or remains critical.

closed 2017-01-04 by edrumwri