RobotLocomotion/drake #2972: Failing tests when Bullet is not installed

jamiesnape, 2016-07-28:
* bullet_collision_zero_rad_sphere_test
* rigid_body_tree_test
* testConstraintConstructor
* testKinematicsCacheChecks
* urdf_parser_test

https://drake-cdash.csail.mit.edu/viewTest.php?onlyfailed&buildid=63218

jwnimmer-tri, 2016-07-28:
The trick here will be disabling tests that require a collision library when no library is present.  We have to be careful that we never accidentally turn off these tests when we thought we wanted them to be testing something.

amcastro-tri, 2016-07-29:
An interesting case here is [`urdf_parser_test`](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/test/urdf_parser_test.cc#L16). This simply tests the parser for a [urdf file that doesn't even have collision elements](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/test/urdf_parser_test.cc#L16). The test however fails because currently the RBT [**always** allocates a collision model](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/RigidBodyTree.cpp#L70) that defaults to a Bullet back end. 
Disabling the test altogether does not seem a good solution since in this case collision is not even used. 

Would a good solution be to delay the allocation of the collision model to when the user tries to add a collision element for the first time? ([here](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/RigidBodyTree.cpp#L345))
If we do so a test like this one could still run without bullet. The only tests that'd fail would be those that do actually require Bullet and the we would of course disable those.

Thoughts?

jwnimmer-tri, 2016-07-29:
My intuition says that the way to approach this issue is to find a collision engine that we can always have available within Drake -- in the same way that there is a simple, default solver always available in `MathematicalProgram.cpp` even if nlopt and ipopt and snopt and gurobi and etc are absent.  For a default collision engine, either write our own slow, naive one, or we find a satisfactorily light-weight one to make required (in `thirdParty`, or a required external).

sherm1, 2016-07-29:
When Sean arrives in September we'll be able to make some serious progress on building our own collision engine. I don't know any collision engine lighter weight than FCL, but even if we build ours on FCL it will take a while to reach minimally-acceptable functionality (that will be the first goal). It's not obvious to me that we would want to consider that as part of Drake rather than an external, although we could reasonably make it required.

In the meanwhile I think it is reasonable to require at least one collision engine, and currently Bullet is the only one that gets the job done. It is completely open source, well supported, and widely used -- what is the advantage of *not* requiring it for now?

jwnimmer-tri, 2016-07-29:
Drake is more than a dynamics simulatior.  The separate optimization framework is very nice.  It would be advantageous to use that without having to supply `Bullet` or `FCL` first.  In general, Drake has a goal of being as flexible and lightweight as possible.  We require `Eigen` and `googletest` (and possibly soon `Gflags`), but allow all other domain-specific software to be optional.  In general, I am more comfortable having required dependencies that are known, used, and tested worldwide (as those three are), and less comfortable having niche items as required dependencies (optimizers, solvers, special-purpose geometry, colliders, etc.).

In any case, if Drake is supposed to only work if you choose >= 1 collision engines as dependencies, then we should have the `CMakeLists.txt` enforce that fact.  Or if we don't want to do that, perhaps having these half-dozen tests loudly disabled when zero engines are selected would be a satisfactory compromise.

sherm1, 2016-07-29:
Bullet also meets the "known, used, tested worldwide" criteria (as well as unrestricted open source and well supported). It is big, though and provides narrow functionality that we want to replace.

jwnimmer-tri, 2016-08-02:
It sounds like it's unlikely that a "just naively compute the collisions with a simple 10-line function" is not an option, unlike with the optimizers.  In which case either:
- (1) We require our users to compile at least 1 collision engine (Bullet or FCL) even if they don't want it, and fail-fast at configure-time if they don't have any, or
- (2) We disable chunks of Drake when no collision library is present, e.g., nothing in `systems` for example.

liangfok, 2016-08-02:
+1 for (2).

jwnimmer-tri, 2016-08-02:
The failing tests are all within `systems/plants`.  That seems like a reasonable thing to disable, though then the challenge is that many `examples` will not work.  (I think we are lucking out on this in terns of #2962 builds, because disabling LCM disables many examples.)

jhoare, 2016-08-03:
Right now I'm working on addressing this by setting a ``HAVE_COLLISION_LIBRARY`` flag which will be a logical OR of every collision library that we support (for now it would be equivalent to if bullet is available) and disable these tests if a collision library is not present. 

Thoughts? I can also look at disabling certain parts of drake using this flag.

jhoare, 2016-08-03:
After looking at this further, I see @amcastro-tri point about a number of these tests are not even performing collision detection/checking, and do not make sense to be disabled. 

Instead, I'm going to implement a Stub Collision Model which will only blow up once it is attempted to be used.

jwnimmer-tri, 2016-08-03:
@jhoare In person here a few days ago, @amcastro-tri had some thoughts on a delayed-construction collision engine selection that would solve this.  You two might want to have that discussion before moving forward.

FWIW, this ticket is not "Priority: high" to me.  Given the difficulty, we can have Bullet be required until a more permanent rework is completed.

jhoare, 2016-08-03:
Well, I had already went ahead and did it and opened a PR #3040. It won't hurt my feelings if we decide that we want to go a different direction.

commit 9034c1e0 (2016-08-04): Merge pull request #3040 from jhoare/minimal_build_fixup

Fix #2972 by adding UnusableModel Collision model
commit 925eb083 (2016-08-04): Fix #2972 by adding UnusableModel Collision model

  * For when Drake is compiled without a collision library (e.g. minimal
    configuration)
  * Will call DRAKE_ASSERT and die when it is attempted to be used.
closed 2016-08-04 by jwnimmer-tri