RobotLocomotion/drake #2974: WARNING "Could not determine MATLAB JVM version"

a137755952, 2016-07-29:
[image]
I found the code,it shows that
    # Ask matlab for its JVM version
    execute_process(
      COMMAND ${MATLAB_EXECUTABLE} ${_args} -r "version -java,quit"
      RESULT_VARIABLE _result
      ERROR_VARIABLE _output
      OUTPUT_VARIABLE _output
      TIMEOUT 180)

    # Test for a valid result
    if(NOT _result AND _output MATCHES "Java ([0-9]+\\.[0-9]+)\\.[0-9_.]+")
      set(MATLAB_JVM_VERSION ${CMAKE_MATCH_1} CACHE INTERNAL "")
      set(_jdk_version "${Java_VERSION_MAJOR}.${Java_VERSION_MINOR}")

      # Compare against JDK version
      if(NOT MATLAB_JVM_VERSION VERSION_EQUAL _jdk_version)
        message(WARNING
        "MATLAB JVM version (${MATLAB_JVM_VERSION}) does not match "
        "installed Java Development Kit version (${_jdk_version})")

        # If JDK is newer, set build options to build Java code compatible
        # with the matlab JVM
        if(MATLAB_JVM_VERSION VERSION_LESS _jdk_version)
          set(CMAKE_JAVA_COMPILE_FLAGS
            -source ${MATLAB_JVM_VERSION} -target ${MATLAB_JVM_VERSION}
            CACHE INTERNAL "")
        endif()
      endif()
    else()
      message(WARNING "Could not determine MATLAB JVM version")
    endif()
  endif()
What should I do next?

a137755952, 2016-07-29:
[image]
This pic was what MATLAB showed during Configuring.

hongkai-dai, 2016-07-29:
BTW, those Chinese characters in the image mean "directory does not exist"

a137755952, 2016-07-29:
when I download drake-distro from git,there are many empty files in file "externals",whether should I download those externals files from git one by one? Still I do not know why cmake could not determine MATLAB JVM version.

a137755952, 2016-07-29:
After Cmake-gui configuring and generating,In the [build artifacts directory](.../drake_build),I can find those files which was not exist before configuring and generating.

liangfok, 2016-07-29:
Drake recently underwent some major changes to its build system and build procedure. Did you try using a fresh clone of the [upstream repository](https://github.com/RobotLocomotion/drake)?

a137755952, 2016-07-29:
I download drake from git shell yesterday @liangfok

jwnimmer-tri, 2016-07-29:
Look at the pasted output.  The problem is that our CMake code is parsing the Java version using `"Java ([0-9]+\\.[0-9]+)\\.[0-9_.]+")`, but the reported Java version has a `b` in it, which defeats our regexp.

jwnimmer-tri, 2016-07-29:
Actually I wonder if the version string is just fine -- it doesn't seem to be anchored, so the `b` part would be safely ignored.  Maybe matlab was returning a non-zero resultcode?

jwnimmer-tri, 2016-07-29:
Actually, the [latest windows matlab Jenkins build](https://drake-jenkins.csail.mit.edu/view/Continuous/job/windows-msvc-64-continuous-matlab/313/consoleFull) also says the same thing:
```
17:24:50 CMake Warning at C:/Users/Administrator/TAa6z2HE/cmake/config.cmake:58 (message):
17:24:50   Could not determine MATLAB JVM version
```

jwnimmer-tri, 2016-07-29:
@a137755952 Are you actually having any problems with your build or using Drake, or were you just calling our attention to the warning?

a137755952, 2016-07-30:
[image]
I do not know whether the warning could call the error as the pic shows after this code "cmake --build . --target ALL_BUILD --config RelWithDebInfo". My classmate had no error and no warning after cmake-gui configuring. @jwnimmer-tri

a137755952, 2016-07-30:
[image]
after typing "ctest -VV -C [build mode]"

liangfok, 2016-07-30:
Does it work if you disable MATLAB?

```
cd drake-distro/build
cmake .. -DDISABLE_MATLAB=TRUE
make
```

Your screenshot above indicates a problem downloading `drakeSnopt.zip`.

a137755952, 2016-07-30:
eh...did you mean "cd ...drake_build"  I can not find the "build"folder in drake-distro @liangfok

a137755952, 2016-07-30:
should I create a new folder "build" in drake-distro @liangfok

liangfok, 2016-07-30:
Oops, it looks like the Windows installation instructions need to be updated. Yes, create a "build" directory within your "drake-distro" directory. I recommend that you first start over with a clean clone:

```
git clone git@github.com:RobotLocomotion/drake.git drake-distro
cd drake-distro
mkdir build
cd build
cmake .. -DDISABLE_MATLAB=TRUE
make
```

Lucy-tri, 2016-09-07:
I think I’m running into the same thing. It’s just a warning, it builds and runs ok. However if you run the gui (ccmake ..) then you get the warning there and have to press “e” to exit that display. 

I created a clean Drake repo on **Ubuntu 16**. With the Matlab cmake option set to build with Matlab (DISABLE_MATLAB=OFF), I got: 
```MATLAB JVM version (1.7) does not match installed Java Development Kit
   version (1.8)```
I ran the build again, and got the warning that the original poster had: 
```Could not determine MATLAB JVM version.```